opentx/opentx #4922: X7 - Crossfire constantly telemetry warning

lichtl, 2017-05-12:
Had an interesting behaviour on a X7 from a friend. Seems that Crossfire telemetry generate constant report telemetry lost/regained.

X7 running RC18 (no-heli, lua, diy module)

The behavior is similar to described in. 
http://fpvlab.com/forums/showthread.php?42097-TBS-CROSSFIRE-Next-Generation-Long-Range-R-C-System&p=832994&viewfull=1#post832994

Crossfire module run on latest stable but even on the beta behavior is the same. This seems once the receiver connect (crossfire telemetry is enabled). Module works fine in X9D plus. PPM also run error free on X7

bsongis, 2017-05-20:
No connection problem? Mine doesn't fit well, I have no contact between the X7 and the module ...

bsongis, 2017-05-22:
I just tested with XFire version 1.63
X7 with OpenTX 2.2.0
Everything runs perfectly

Which XFire version are you testing?

lichtl, 2017-05-25:
We tried 1.63 and latest stable (think 1.57). Will gonna try to bend the pins a bit more (maybe only TLM pin) do not have proper connect.

bsongis, 2017-05-25:
yes I think it's a bad connection

stevef1966, 2017-06-02:
Hi im having the same issue, telemetry looping, connection seems fine to the crossfire, everything is on latest recommended firmware

stevef1966, 2017-06-02:
dont know if this is the correct way to do this but....

I have re-flashed all firmware (yet again) and checked for loose pins/bad connections to the crossfire TX still get the telemetry lost/recovered issue, have tried to use the PPM protocol in openTX to talk to the crossfire TX and this is the only way to stop the constant telemetry warnings, is this a bug?

bsongis, 2017-06-02:
I will provide a test firmware for you so that you will collect traces as a basis for investigation

stevef1966, 2017-06-04:
thanks, look forward to testing

stevef1966, 2017-06-06:
hi did you get a chance with the firmware?

bsongis, 2017-06-10:
Here:
http://downloads.open-tx.org/2.2/firmware/tests/firmware-x7-telemetry-logs.bin

bsongis, 2017-06-10:
You will have a new file telemetry.log on your SD. This will be intesting

stevef1966, 2017-06-11:
Awesome do I just upload the file back here?

stevef1966, 2017-06-11:
[telemetry.log.txt](https://github.com/opentx/opentx/files/1066348/telemetry.log.txt)

stevef1966, 2017-06-11:
put the firmware on the qx7, turned on and changed from PPM on the radio to CRSF, the looping starts at that point, moved sticks and switchs to confirm connection with receiver via betaflight.

bsongis, 2017-06-11:
perfect, I will investigate when I have one minute, I prepared a script for this

bsongis, 2017-06-11:
we really miss a lot of bytes! do you have complex scripts, SD logging or a lot of functions running in parallel?

stevef1966, 2017-06-11:
i have nothing running, just a basic model, have not setup any functions other than a count down timer on the main arming switch (SF i think), dont believe SD card logging is turned on at all, but i can check just to be on the safe side

bsongis, 2017-06-11:
if you disconnect everything from your XF Rx is it better? just to remove betaflight from the equation...

stevef1966, 2017-06-11:
just disconnected the TX/RX pins (so the FC just provides power) from betaflight and its the same

stevef1966, 2017-06-11:
do you want me to do another test without the flight controller in and send the logs, see if you can see anything different?

bsongis, 2017-06-11:
yes please!

stevef1966, 2017-06-11:
will upload tomorrow if thats ok

stevef1966, 2017-06-16:
hows it going, did you find anything new?

bsongis, 2017-06-17:
I still see a lot of [Attitude] / [Flight Mode] fields, this seems weird to me as you said you disconnected all devices from your RX

bsongis, 2017-06-17:
[telemetry-decoded.txt](https://github.com/opentx/opentx/files/1082402/telemetry-decoded.txt)

bsongis, 2017-06-17:
Almost all frames are received with a wrong CRC

bsongis, 2017-06-17:
obviously, and provided the wiring is good it's a bug on OpenTX side

bsongis, 2017-06-17:
and the wiring is good, you receive channels on the CC and channels are transmitted via the same S.PORT PIN

stevef1966, 2017-06-17:
i believe so, i will double check, previously i'd even re-soldered the internal qx7 pins just in case, i've even made a small harness to connect the qx7 pins to the crossfire to test the pins LOL... 
i'll reconnect and check today and let you know..

stevef1966, 2017-06-17:
just tested again, tried to jiggle the pins, the only place (if its a pin/connection problem) is inside the crossfire itself, might try and see if i can open it up

bsongis, 2017-06-18:
I don't believe it's possible, you wouldn't receive the channels

bsongis, 2017-06-18:
I have Crossfire V1.63 (beta account)
And my logs look like this, no CRC error ... 
```
[2000-01-01,00:04:06.220] ea 0c 14 1f 1f 64 56 ff 02 01 17 64 5f 87 [Link Statistics] 
[2000-01-01,00:04:06.220] ea 0c 14 1f 1f 64 56 ff 02 01 17 64 5f 87 [Link Statistics] 
[2000-01-01,00:04:06.230] ea 0c 14 1f 1f 64 57 ff 02 01 17 64 5f 13 [Link Statistics] 
[2000-01-01,00:04:06.230] ea 0c 14 1f 1f 64 57 ff 02 01 17 64 5f 13 [Link Statistics] 
[2000-01-01,00:04:06.240] ea 0c 14 1f 1f 64 57 ff 02 01 17 64 5f 13 [Link Statistics] 
[2000-01-01,00:04:06.240] ea 0c 14 1f 1f 64 57 ff 02 01 17 64 5e c6 [Link Statistics] 
[2000-01-01,00:04:06.240] ea 0c 14 1f 1f 64 57 ff 02 01 17 64 5e c6 [Link Statistics] 
[2000-01-01,00:04:06.250] ea 0c 14 1f 1f 64 57 ff 02 01 17 64 5e c6 [Link Statistics] 
[2000-01-01,00:04:06.250] ea 0c 14 1f 1f 64 57 ff 02 01 17 64 5e c6 [Link Statistics] 
[2000-01-01,00:04:06.260] ea 0c 14 1f 1f 64 55 ff 02 01 17 64 65 4e [Link Statistics] 
[2000-01-01,00:04:06.260] ea 0c 14 1f 1f 64 55 ff 02 01 17 64 5b 10 [Link Statistics] 
[2000-01-01,00:04:06.260] ea 0c 14 1f 1f 64 55 ff 02 01 17 64 5b 10 [Link Statistics] 
[2000-01-01,00:04:06.270] ea 0c 14 1f 1f 64 55 ff 02 01 17 64 5b 10 [Link Statistics] 
[2000-01-01,00:04:06.270] ea 0c 14 1f 1f 64 55 ff 02 01 17 64 5b 10 [Link Statistics] 
[2000-01-01,00:04:06.280] ea 0c 14 1f 1f 64 55 ff 02 01 17 64 5b 10 [Link Statistics] 
[2000-01-01,00:04:06.280] ea 0c 14 1f 1f 64 59 ff 02 01 17 64 5b 61 [Link Statistics] 
[2000-01-01,00:04:06.280] ea 0c 14 1f 1f 64 59 ff 02 01 17 64 5b 61 [Link Statistics] 
[2000-01-01,00:04:06.290] ea 0c 14 1f 1f 64 59 ff 02 01 17 64 5b 61 [Link Statistics] 
[2000-01-01,00:04:06.290] ea 0c 14 1f 1f 64 59 ff 02 01 17 64 5b 61 [Link Statistics] 
[2000-01-01,00:04:06.300] ea 0c 14 1f 1f 64 59 ff 02 01 17 64 5b 61 [Link Statistics] 
[2000-01-01,00:04:06.300] ea 0c 14 1f 1f 64 59 ff 02 01 17 64 5c 35 [Link Statistics] 
[2000-01-01,00:04:06.300] ea 0c 14 1f 1f 64 59 ff 02 01 17 64 5c 35 [Link Statistics] 
[2000-01-01,00:04:06.310] ea 0c 14 1f 1f 64 52 ff 02 01 17 64 5c 02 [Link Statistics] 
[2000-01-01,00:04:06.310] ea 0c 14 1f 1f 64 52 ff 02 01 17 64 5c 02 [Link Statistics] 
[2000-01-01,00:04:06.320] ea 0c 14 1f 1f 64 52 ff 02 01 17 64 5c 02 [Link Statistics] 
[2000-01-01,00:04:06.320] ea 0c 14 1f 1f 64 52 ff 02 01 17 64 5c 02 [Link Statistics] 
[2000-01-01,00:04:06.320] ea 0c 14 1f 1f 64 52 ff 02 01 17 64 5c 02 [Link Statistics] 
[2000-01-01,00:04:06.330] ea 0c 14 1f 1f 64 52 ff 02 01 17 64 47 d3 [Link Statistics] 
[2000-01-01,00:04:06.330] ea 0c 14 1f 1f 64 52 ff 02 01 17 64 47 d3 [Link Statistics] 
[2000-01-01,00:04:06.340] ea 0c 14 1f 1f 64 59 ff 02 01 17 64 5f 9f [Link Statistics] 
[2000-01-01,00:04:06.340] ea 0c 14 1f 1f 64 59 ff 02 01 17 64 44 4e [Link Statistics] 
[2000-01-01,00:04:06.340] ea 0c 14 1f 1f 64 59 ff 02 01 17 64 44 4e [Link Statistics] 
[2000-01-01,00:04:06.350] ea 0c 14 1f 1f 64 59 ff 02 01 17 64 63 be [Link Statistics] 
[2000-01-01,00:04:06.350] ea 0c 14 1f 1f 64 59 ff 02 01 17 64 63 be [Link Statistics] 
[2000-01-01,00:04:06.360] ea 0c 14 1f 1f 64 59 ff 02 01 17 64 43 1a [Link Statistics] 
[2000-01-01,00:04:06.360] ea 0c 14 1f 1f 64 57 ff 02 01 17 64 64 66 [Link Statistics] 
[2000-01-01,00:04:06.360] ea 0c 14 1f 1f 64 57 ff 02 01 17 64 64 66 [Link Statistics] 
[2000-01-01,00:04:06.370] ea 0c 14 1f 1f 64 57 ff 02 01 17 64 63 32 [Link Statistics] 
[2000-01-01,00:04:06.370] ea 0c 14 1f 1f 64 57 ff 02 01 17 64 63 32 [Link Statistics] 
```

stevef1966, 2017-06-21:
do you think its the radio or the crossfire unit thats at fault? tell me what you need me to do, think i might start from scratch and try to eliminate everything where possible

stevef1966, 2017-06-21:
just had the crossfire module bay adaptor in bits, have reworked the connections on the board... still the same, not looking like a physical hardware issue on my radio..

bsongis, 2017-06-21:
do you confirm you use Crossfire V1.63?

stevef1966, 2017-06-23:
currently im on the latest version, think its 1.65, i have tried all versions, i can go back to that specific version if you like, if it helps

bsongis, 2017-06-25:
I just received a new production X7 and I reproduce your problem, I need now to investigate further

bsongis, 2017-06-25:
Same firmware on the 2 radios, almost the same curve on the S.PORT Pin, same level. I need now to know if something changed recently (which would remove the possibility of going up to 400kBauds)

stevef1966, 2017-06-26:
Good news, in a weird way I guess, if I can help further please let me know

anthonycake, 2017-06-27:
I can confirm this behavior. A crossfire taken from an 'old' Taranis, with telemetry working fine, then plugged into a Taranis X7, has an extremely poor telemetry link. 
Looking with an oscilloscope on the S-PORT pin shows a beautiful 400k baud serial stream in both directions.

stonefoxfpv, 2017-06-29:
Also having the same problem, I had the Crossfire module installed into my X9D Plus and worked fine, on latest bata's was able to see telemetry from KISS FC from Crossfire on the Taranis. Now I've installed it on the QX7 and it seems to be recieveing telemetry but constantly says telemetry recovered/lost. Insterestly it does it when the quad isn't powered on. My brother is also having same issue. 
I did notice there is a folder called Crossfire in  the X9D Plus SD card files, but no folder in the X7 SD card files. Which seems odd to me...

bsongis, 2017-06-29:
The crossfire receiver has an internal battery, it continues to send telemetry once powered off

stonefoxfpv, 2017-06-29:
Sorry, I'm using the Crossfire micro RX, which I don't think has a battery. 
Do you know why there isn't a Crossfire folder on the X7?

bsongis, 2017-06-29:
Right for the micro RX. OpenTX has also a buffer for prompts to be played, so your buffer is full of "tememetry lost" / "telemetry recovered". Makes me think that we could do better, but this is not the issue we are discussing here

The crossfire script needs to be adapted for the small X7 screen. I will try to take care of this, but once we have a correct link ...

stonefoxfpv, 2017-06-29:
Oh, gotcha. That makes sense. Thanks!

stevef1966, 2017-07-24:
Hi any news on a fix for this?

Drewlrox, 2017-07-25:
Was about to order the new smaller crossfire module then I saw this, anyone find any fixes? Would deleting the files work? Or could that cause the radio to freeze with all the requests to play the sound?

Kalored, 2017-07-27:
Hi, I have the same issue on my QX7 as written above. 
With CRSF enabled. Would love to see a fix to that as it is crucial to me to have Telemetry packages. 
PPM also only have 8 channels which are a little too less. Running also with MicroRX v2

Drewlrox, 2017-07-28:
Have you tried deleting the telemetry voice files and flying that way? If so did it work fine?

Drewlrox, 2017-08-01:
So I have a crossfire micro pre ordered, somebody told me erasing the voice files has been working fine for them. Is a lua script being developed? The micro tx has no screen and relies on the lua script to change settings.

lichtl, 2017-08-01:
Removing the files works. I just don't think thats the right way since this also remove warning on internal module. I think it has been proven that there is something wrong with the serial link.

Drewlrox, 2017-08-09:
Any updates? My crossfire micro is on the way but I cant find a lua script still.

lichtl, 2017-08-10:
You can try copy the CROSSFIRE folder from X9 SD on X7, maybe it work.

Drewlrox, 2017-08-10:
It it runs but cuts off the values on the right side, I've spent lots of time trying to modify the script myself but I know nothing about lua so I havnt made it very far.

OhmEye, 2017-08-11:
I'm looking for a solution as well, new X7 flashed with latest release and models copied from my X9D. In the X9D the crossfire works perfectly and steady telemetry. With the X7 the telemetry is rapidly being lost and recovered with voice constantly alternating between "telemetry lost" and "telemetry recovered" . When it says no telemetry there is indeed no telemetry. Just disabling the voice isn't what I want, I want to be able to receive and log telemetry.

No lua scripts, nothing very complex running on either my X9D or X7, both running current 2.2.0 release with same options. Control works normally on both, only telemetry on the X7 is flapping like mad.

bsongis, 2017-08-11:
The X7 is not able to communicate with the Crossfire module. The speed is too fast. We have to add the possibility of downgrading to 200kBauds instead of 400kBauds we have on X9D / X9E.

Drewlrox, 2017-08-12:
Is that something that is possible with software? if it was done do you think tbs would be able to support this through software as well?

kilrah, 2017-08-12:
Yeah it means both TBS and us need to decide on another speed and implement settings to choose which to use.
What worked was an early X7, the one without the SPORT connector at the bottom... but it isn't clear if it was just a stroke of luck that this one worked or if that whole series would have.

Drewlrox, 2017-08-12:
Hmmm, My radio bottom sport works, Do you think its possible if i were to solder a wire form the sport pin on crossfire to the sport on the bottom of the radio it could work? Or are they completely separate?

aplikasiclient, 2017-08-15:
-subs-
i have a same problem, waiting for solve this problem

[200 more replies]

commit 390342af (2017-09-10): Crossfire baudrate selection for X7 support ( fixes #4922 ) (#5194)

* Crossfire baudrate selection for X7 support ( fixes #4922 )

* Moved baurate selection to general settings

* renamed mavbaud to telemetryBaudrate for ARM targets
translation added

* typo fixed
closed 2017-09-10 by bsongis