opentx/opentx #4928: In mixer, slow works, delay not

nostromo123, 2017-05-13:
Using 2.1.8 with Taranis X9D. Trying to add some functionally with a flying wing.

Which is the slow/delay addition in mixer. 

Slow function works on model and in simulator.
Exchanging slow by delay, no function.

Tried Companion   2.2.0N371 and see same behavior. Slow YES, delay NO.

Please have a look. eepe file added.

br

[documentx.eepe.zip](https://github.com/opentx/opentx/files/998813/documentx.eepe.zip)

nostromo123, 2017-05-13:
Made a short video. Hope, this explains better ....
Stars with SLOW on CH 2 and CH 3. Then replaced SLOW on CH 2 with DELAY. No function with delay.

https://youtu.be/-54_KSg77PI

kilrah, 2017-05-13:
It looks like Delay does not work if a switch is set. Not sure if it's intended or not.

kilrah, 2017-05-13:
If a switch is set then delay operates at the activation/deactivation of the line instead of a source value change.

nostromo123, 2017-05-13:
As I understand documentation, SLOW and DELAY should be handled in the same way ...

Nihonski, 2019-07-30:
I want to report the same problem. See Screenshot.
Channal 30 Delay does not work. Channal 31 Slow works.
Petty, I hope you can solve the issue for OTX 2.2 and 2.3
[image]

Nihonski, 2019-07-30:
kilrah: "If a switch is set then delay operates at the activation/deactivation of the line instead of a source value change."
This funtions does not have any practical usage. 
Please change into same way as SLOW

kilrah, 2019-07-30:
It is not possible to delay an analog value in any sensible way without unreasonable ressource usage.

Nihonski, 2019-07-30:
After giving a second thought, to delay an analog signal would need some kind of recording and replay. Understand, not that easy, would need quite some memory.
How about not delay the analog signal, but delay the slow function? Means the slow funtions would start instantly or after a defined delay.

Nihonski, 2019-07-30:
I developed an fuel injection system which uses two channels, one is throttle, the other is fuel pump.
When pushing the throttle stick, the fuelpump should speedup slightly ahead in time before the carburator opens. 
In case of real analog signal delay, I would need not more than 0.3s

Nihonski, 2019-07-30:
Thank you for your quick response.
Presently the Slow functions is a variable slope.
 
How about the implement a variable delay without any action before starting the slope.

Nihonski, 2019-07-30:
I checked again the pesent OTX 2.2.4
The delay - slow - function works fine as I expect. 
The problem is only when a switch condition is activated.
Please check.

kilrah, 2019-07-31:
> When pushing the throttle stick, the fuelpump should speedup slightly ahead in time before the carburator opens.

Then you can just use a logical switch with a delay to enable the carburator control.

Nihonski, 2019-07-31:
The way how "delay" and "slow" presently works is fine.
The Problem is the conditional switch.
It is not understandable, why the delay function does not work in the same way with a switch, as without a switch. The switch should only have the function of enable or disable the respective mixer.
I think the switch in a mixer is mistakely used as a trigger for the delay.
Delay should be triggered whenever there is a Change on the analog signal.
Delay should freeze the analog Signal for the specified time, and then continue with the actual value.
If delay and slow is combined, then the signal should freeze first and then ramp up to the present analog value.
This is works presently perfect without a switch. 
Please use the switch only for enable/disable the respective mixer line.

kilrah, 2019-07-31:
The current behavior is known and has been that way for many years. We cannot just change this when >100'000 people expect it to work the way it works.

Nihonski, 2019-07-31:
I dont think that anybody expects and uses the delay in combination with a switch in the way as it is. At least not in the FrSky-Forum.
Anyway it is your decision. To finilize this, could you let me know any practical example how the present behavior is used in real life.

closed 2020-04-26 by stale[bot]