opentx/opentx #4935: Companion 2.2: rotenc firmware option missing in companion for Mega2560 radio

hjuergen, 2017-05-18:
The checkbox for rotenc option seems to be missing for DIY Mega2560 radio.

bsongis, 2017-05-18:
shouldn't it be always enabled?

hjuergen, 2017-05-18:
should be fine to always enable rotencs. Looked in opentxinterface and did not find any nav_options for Mega2560 board. Thus I concluded rotencs are not enabled. However, I am ignorant in programming, so I might be wrong and instead am facing a hardware problem that rotencs don't work in my setup.

bsongis, 2017-05-22:
[image]

bsongis, 2017-05-22:
Don't you have those 2 rotary encoders REa and REb?

hjuergen, 2017-05-23:
Your snapshot helped: in switches menue I see REa and REb, I could invert zeros by pushing rotary encoders, but no change in numbers by rotating. Finally found hardware issue (crossed cables). After enabling rotary encoders for navigation in radio setup menu I can now use encoders for navigation (sometimes however with jumping over one or two lines). Thanks again. Please close this issue.

3djc, 2017-05-23:
Thx for the feedback!

closed 2017-05-23 by 3djc