opentx/opentx #4941: Companion "print model" function shows non existent input lines

strgaltdel, 2017-05-22:
Hi,
got an Horus otx file from a friend who discovered a minor "problem"
The "print model" option shows input lines which were deleted in the past,
(Maybe converted lines from a long ago nightly?) and so didn't exist in the real model.
Another option could be a corrupted  / messed up file?
No error by opening this file.

Have tested this file back to nightly 362, problem still exists.
When i add a line in the affected inputs, printout shows two lines.
I wasn't able to delete these "foo lines"
File download
http://www.filedropper.com/hybrid

strgaltdel, 2017-05-22:
[image]

mpaperno, 2017-05-22:
I think the underlying issue might be that the input lines aren't getting properly deleted in the first place.  I'm not sure why.  But with other models and even other inputs in the "HYBRID1" model when I delete them (r-click -> Delete) the whole line goes blank including the input name.  But all the existing input lines can't be deleted properly. Have to investigate further later on, or maybe someone else knows what I'm missing here.

3djc, 2017-05-22:
We would be interested in learning more on how (and when) the file/model was created (imported from another radio?). We believe current version of companion and radio firmware do not allow this (unclean entries)

mpaperno, 2017-05-23:
Turns out the model printer had it right.  Both Companion and radio code are written such that anything after a "blank" input (`model->expoData[i].mode == INPUT_MODE_NONE`) is not displayed in the Inputs list (except for the custom name, if any).  Or in other words it assumes the `expoData[]` definitions are all consecutive, which they should be if items are being properly created/deleted.  (I will submit a patch for Companion which shows the "phantom" input lines and can properly delete them.)

This doesn't explain why the inputs were not deleted correctly here in the first place.  I cannot recreate that problem in current Companion code.  I haven't tested from radio side.

I suspect the Mixes display has the same issue.

strgaltdel, 2017-05-23:
I've requested more detailed infos how the lines were created , which nightlys "passed" this otx file and so on.
Guess today i will get an answer

strgaltdel, 2017-05-23:
Only info i got was, that it's very unlikely the config was initiated by 2.1.8. / transformed to 2.2
As far as he remembers the file was created some months ago.

From what i know about the user is, 
- he is a formerly X9e owner and switched to horus 2month ago
- he is not using win10
- he never used wizzards

sorry i can't provide more infos...

3djc, 2017-05-24:
We believe this corrupted model situation is a result of old bugs that have been fixed since, and an very unusual set of circumstances, and that with today's code, you should not get into that situation. 

At this point in time, we do not intend to do anything specific to help cleaning those models, they will have to be recreated manually, sorry about that.

If a number of similar instances appears, we will obviously review the situation

closed 2017-05-24 by 3djc