opentx/opentx #4943: Companion model edit corrupt mixes list when adding mix to line 1 of a channel

elecpower, 2017-05-23:
Adding a mix line to line 1 of a channel temporarily corrupts the Mixes list. Adding to a line 2 onwards does not exhibit the issue. Exiting the mix edit dialogue refreshes mixes list correctly.

[image]

Companion 2.1.9 Win 10 64bit Taranis X9D+

mpaperno, 2017-06-03:
This is fixed in #4919.  (I noticed it while working on that PR but thought it was something I had caused in the new code, so wasn't aware it was an issue to begin with.   Not entirely sure I remember the original issue, eg. for back-porting to 2.1.10.)

elecpower, 2017-11-16:
confirm fixed at 2.2

closed 2017-11-16 by elecpower