opentx/opentx #4944: Companion model edit cancelling add of input line leaves skeleton

elecpower, 2017-05-23:
Highlight an input line, from context menu select Add, click Cancel from edit dialogue.

[image]

Companion 2.1.9 Win 10 64bit Taranis X9D+

mpaperno, 2017-11-12:
Perhaps more an enhancement than bug... When you use Add it properly adds a line.  Then, for the user's convenience the edit dialog is automatically opened on that line.  To "fix" this we'd need to determine that Cancel was pressed on a brand new line which was just added, and only then delete that line.  I'm sure it's possible and probably even logical, but it has to be coded properly to work w/out unwanted side-effects.

mpaperno, 2017-11-12:
NVM, does seem like a bug... should have looked at the code first.  There's code in `modeledit/inputs.cpp` which should already delete a new input if dialog was cancelled, but it doesn't work for some reason.  Does work in mixers list.

PR #5397 (2017-11-23): Fixes 4944 Skeleton input after cancel of add
Fixes #4944
closed 2017-11-23 by bsongis