opentx/opentx #4949: Companion Edit Windows

CraigHB, 2017-05-23:
In Windows OS Companion does not provide a hot key for switching model edit windows.  Most programs allow the use of ctrl->tab to switch windows within a program.  If a window covers another window there's no way to get to it without closing the window in the foreground.  Also would like to compare data in edit windows the same size and position by switching between them.

mpaperno, 2017-05-25:
> In Windows OS Companion does not provide a hot key for switching model edit windows. 

AFAIK Windows doesn't provide this functionality (switching between top-level windows of an app) by default.  On Mac one can use `Cmd+~`.  You could try a third-party utility such as:  https://neosmart.net/EasySwitch/ or  http://www.ntwind.com/software/vistaswitcher.html

You could also run multiple instances of Companion and switch between them using Alt-Tab.

> Most programs allow the use of ctrl->tab to switch windows within a program.

This switches between open file windows inside the main application window.  Does not work for the editor dialogs though.

> If a window covers another window there's no way to get to it without closing the window in the foreground.

Can't you move it out of the way or resize the top one so other windows are visible?

The only real alternative that comes to mind would be an option to display the editor windows inside the main window (as part of the MDI).  Which then would allow tabbing/arranging and Ctrl-Tab switching.  But I'm not sure everyone would enjoy that, personally I like them floating outside the main app window.

CraigHB, 2017-05-25:
Floating windows (independent instances) is fine, but there's no task icon to switch between edit windows either.  That would be fine for me if that were the case.  Yes I can open multiple instances of Companion to alt-tab or click taskbar icons to switch between windows so that is a work-around for me.  However, very easy to get confused and start editing two different instances of the same model.  My feeling is the program should either open edit windows as a wholly separate instance and create a task icon for each instance or operate within the main program window with a hot key to switch between them.  At this point I feel it does neither properly.

mpaperno, 2017-05-25:
>  Yes I can open multiple instances of Companion... However, very easy to get confused and start editing two different instances of the same model.

No different than using one instance of Companion, since opening the same model multiple times is allowed.  Whether that's a bug or a feature is debatable. 

> ... the program should either open edit windows as a wholly separate instance and create a task icon for each instance...

Yes that is actually easy to do (Win/Linux taskbar entry for each editor dialog). But the side effect is that when you close the main window, any editor dialogs remain open. They need to be closed individually, and more importantly there's no way to save any changes once the main window is closed. Awkward at best. So we'd need our own mechanism to close any editors when main window is closed... which should be relatively simple.

> ... or operate within the main program window with a hot key to switch between them.

That would be the alternative I mentioned first (editor windows inside the MDI). I could see how it may be nice on smaller screens where it is only practical to view one window at a time anyway. 

> At this point I feel it does neither properly.

To be fair, it doesn't try right now to do either.  😉   Not sure anyone's brought this up before.

CraigHB, 2017-05-25:
Yes that would be like six of one half dozen of the other.  I actually was thinking, wait, there's no difference since companion allows you to open more than one edit instance of the same model anyway.  Bug or feature is debatable, easy to get confused on that.

I seem to recall another program I use that places a taskbar icon for every window instance, oh yeah, Eagle CAD (for PCB drawing).  It will put up a taskbar icon for each window you open from the main program instance.  Closing the main program also closes all the windows opened from it.

So that would be the solution in my mind, present a taskbar icon for each edit window, then program it so closing the main Companion windows closes all windows associated with it.  If you happen to use Eagle which is a really common program for its purpose you'll see what I mean.

closed 2019-11-15 by stale[bot]