opentx/opentx #4950: Global Persistent Timer

CraigHB, 2017-05-23:
There's no simple way I can find to create a persistent global timer that runs whenever the transmitter is on.  Would like to be able to keep track of how long the radio runs on a charge.  Any persistent timers created using a model revert to the last time recorded with that model active so it can't be done there.  Running Companion on Windows OS, OpenTX version 2.2.0 N371.

kilrah, 2017-05-23:
Radio type?

CraigHB, 2017-05-24:
Taranis Q X7

rdeanchurch, 2017-05-28:
A nice feature.  It could also be used for the total time the Tx has been used.
If there was an EVENT for power off button depressed and maybe a Function script could write this dat to the SD....given some processing time after EVENT _POWER_OFF button detected.  The delay for power off seems like more than enough time would be available.  
This could also be expanded to include a log for each models selected time.

jj604, 2017-05-29:
Kilrah, can I just comment that IMO it would be good to have a "simple" option exactly as CraigHB requested however in addition to any more powerful functionality. Not many users will be into scripting while all might find a "Time since last charge" useful. Both my original DX8 and my 10 year old Cockpit SX have such a manually resettable timer.  Thanks.  John

mpjf01, 2017-05-29:
There presumably is a reason why only the QX7 is indicated as the enhancement target? From the discussion on RCG that prompted the request it would seem that it is desired on all platforms.

kilrah, 2017-05-29:
All platforms already have a global running timer. It's simply not displayed anywhere on the X7 (and other 128x64 ARM platforms) and this is about adding it in the UI. Nothing to do with Lua or anything advanced, please open separate issues for that (there might even already be one).

3djc, 2017-05-29:
There is a timer responding to requirement of : " keep track of how long the radio runs on a charge.". On x9 for example, it is shown in statistics, and can be reset on demand by long menu press.

There is no UI for it (yet) on x7, that's what we will be adding

mpjf01, 2017-05-29:
Thanks for that. BTW "Statistics" comes from long ENTER press. Some of the values shown are a bit cryptic, What is SES for example. After TOT I get a value 15C39:26. How does the timer know when the battery is recharged, is this automatic somehow or does the user have to reset something?

3djc, 2017-05-29:
It is a manual triggered thing

CraigHB, 2017-05-29:
Excellent, is there a way to find out when this will be added so I can make use of it?

3djc, 2017-05-29:
Not before 2.2.1

CraigHB, 2017-05-29:
So do I check a change log on 2.2.1 to find out if it's there?

kilrah, 2017-05-29:
This issue will be closed when it's added.

PR #4985 (2017-06-02): Align Taranis 128*64 stats to its larger sister
[image]

This fixes #4950
closed 2017-06-02 by bsongis