opentx/opentx #4952: Italian time sounds "rendering" is broken

SYNdrone-xx, 2017-05-25:
I'm used to set a button to play TH% Timer during flight (special functions->play value->Timer x).
Voice reading is fine in English but broken in Italian. 
The bug is really easy to reproduce but I'm not able to find a pattern to describe how the time to text conversion fails in Italian.

3djc, 2017-05-27:
We do not speak italian, so we will not be able to fix it unless somebody explains how it works in Italian. Here is an exemple of what was done for Cz : https://github.com/opentx/opentx/blob/2.2/radio/util/tts_doc_cz.md

We will not be able to fix it unless someones details how things should be said in a way we can translate into code.

SYNdrone-xx, 2017-05-27:
Well, I think the problem is not related on syntax or grammar issue.  
It should be more trivial. 
On OpenTX 2.1.x was working correctly.
In example: when timer is 00'30" the "play value" functions says: "Zero seconds", on 01'10" says "and 1 second", on 02'00" says "2 minutes",  on 02'30" says "2 minutes and 2 seconds" and so on.
Anyway I'll try to setup a dev environment and fix the bug myself.

Is 2.2 much different from 2.1 TTS wise?

3djc, 2017-05-27:
Yes, 2.2 units are different, but I would not rate that as too different. I won't have access to my dev env before monday, but given what you describe above, I should be able to fix that on monday

SYNdrone-xx, 2017-05-27:
Ok, thank you very much! 
Let me know if I can help testing or writing some specs about italian syntax/grammar.

3djc, 2017-05-28:
Are you running next or N371? Already did a bunch of fixes in #4889

SYNdrone-xx, 2017-05-28:
As stated in first comment I'm using N371.  Should I compile OpenTX myself to try next or can I find a build to download?

3djc, 2017-05-28:
If you give me a radio type and the option you use, I can compile one for you

SYNdrone-xx, 2017-05-29:
Ok, compiling PyQt4 is driving me crazy so I accept you offer:  Taranis+, ppmus, noheli, lua, bindopt. Thank you very much. Btw, it would be interesting to try Companion as well, mainly for the simulator.

3djc, 2017-05-29:
http://downloads.open-tx.org/2.2/tests/opentx-x9d+-fix-italian.bin

Not meant to be flown, just for language testing

SYNdrone-xx, 2017-05-29:
Tested right now. Seems to work fine! Time values are perfect, tomorrow I'll try other value types.

3djc, 2017-05-30:
I'm closing it, but feel free to report here issues if you find one

closed 2017-05-30 by 3djc