opentx/opentx #4956: Q X7 & iRangX 4in1 & AFHDS 2a = Barometric Telemetry Sensor TGY-CAT01 not found

EvilPusher, 2017-05-27:
**Hardware:** 
Unsing Taranis Q X7 + iRangeX 4in1 Module (IRX4).
**Firmware:**
Version 2.2.0N371 (4df32f82). 

All sensors (2x temp, 1x voltage) are found - except barometric sensor Turnigy **TGY-CAT01** (hobbyking sku 9114000051-0)

I guess, it´s still not implemented?

schwabe, 2017-05-27:
Do you any sensor there? Unknown sensors should at least should be visible with a raw value

nebbian, 2017-08-31:
This has happened to me as well.  OpenTx didn't discover the altitude sensor (however it did discover the pack voltage sensor, which is part of the altitude sensor package).  I tried manually adding a sensor and tried all the IDs between 0x00 and 0x110 without any luck.

schwabe, 2017-08-31:
@nebbian does discover find any other sensor that is unknown? E.g. a snesor that reports just strange numbers?

nebbian, 2017-08-31:
Hi @schwabe, no it did not.  It only found the pack voltage sensor once I'd plugged in the altitude sensor, and asked OpenTx to do a discovery.  This is in addition to the other RSSI type sensors that it found after I'd first bound to the Rx with no sensors plugged in.

nebbian, 2017-08-31:
@schwabe, it appears as if OpenTx is only looking at the sensor IDs defined here:
https://github.com/opentx/opentx/blob/2.2/radio/src/telemetry/flysky_ibus.cpp#L52

Compare that list to this one:
https://github.com/hali9/inav/blob/development/src/main/telemetry/ibus_shared.h#L33

And you'll see a whole lot more iBus goodness that could be used.

schwabe, 2017-08-31:
@nebbian Look further down https://github.com/opentx/opentx/blob/6cc40c65b22ca90302d992ea91ed0577d61ce978/radio/src/telemetry/flysky_ibus.cpp#L109 

If the sensor is not in that table it will add the sensor as RAW sensor. 

But the table in the cpp file is more interesting, but that they use the same type for two sensors 

```c++
    {.type = IBUS_MEAS_TYPE_EXTERNAL_VOLTAGE, .size = 2, .value = IBUS_MEAS_VALUE_CURRENT },          // Address 5, sensor 6, Current in A AS ExtV
    {.type = IBUS_MEAS_TYPE_EXTERNAL_VOLTAGE, .size = 2, .value = IBUS_MEAS_VALUE_ALT },              // Address 6, sensor 7, Baro Alt in cm AS ExtV
```

But even then the sensor should appear as two sensors. Do you know how to use a terminal (or maybe even to compile yourself?). I could send you a version that logs the raw telemetry packet to the usb serial console so I can check what the multi module outputs so we don't have to do guesswork here.

nebbian, 2017-08-31:
Hi @schwabe, yes I'm quite familiar with the terminal.  Happy to run some custom firmware and log the terminal output if required.  I'm currently downloading OpenTx and will install the toolchain as well.
Many thanks for looking at this for me.

schwabe, 2017-08-31:
If you compile it yourself just make a ```#if 1``` instead of ```#if 0```

https://github.com/opentx/opentx/blob/6cc40c65b22ca90302d992ea91ed0577d61ce978/radio/src/telemetry/flysky_ibus.cpp#L143

and then compile with -DCLI and -DUSB=SERIAL. If you tell me what you remote you have I can compile a firmware for you.

nebbian, 2017-08-31:
OK this is embarrassing.  I have a CVT01, not a CAT01.  So I don't actually have an altitude sensor at all!  This is the first time I've used the flysky telemetry system, and it's for a mate, and I asked him to get both sensors, and I thought the one he gave me was the altitude sensor.
Please disregard everything I've written above.

I'll update this issue once I've actually tried the CAT01.

schwabe, 2017-08-31:
@nebbian no problem, we all make mistakes. I still wonder where the list in inav comes from, as far as I know most of these sensors don't exist

nebbian, 2017-09-28:
@schwabe, I recently received a package with the altitude sensor in there.  Sadly the behaviour was as described above.  I do the sensor discovery, and it finds the voltage sensor, but can't find the altitude sensor.  No extra channels are shown as Raw either.  I double checked that this is actually the CAT-01, and yes it is.  It is plugged into the sens port, and the voltage sensor (which IS picked up) is daisychained into the altitude sensor.  Happy to post a video, or run some custom firmware to help to track down the issue.  Alternatively I could post you a sample receiver and sensor for you to work on yourself.

nebbian, 2017-09-28:
Forgot to mention, I'm using a Taranis X9D, with an older style 8 bit Bangood 4 in 1 module if that helps.

schwabe, 2017-09-28:
4in1 avr or stm should not make any difference. I am bit out of ideas on this what to try from a remote diagnosis. If you have a logic analzyer you could listen in between the receiver and the CAT-01 sensor to see if it really transmits its data to the reiceiver. Do you still have a original Flysky remote to check if the sensor works with the original one?

qba667, 2018-04-16:
@schwabe the sensor is using different frame AC (not AA as 2 bytes sensors).

schwabe, 2018-04-16:
@qba667 do you have more information or maybe a dump what it is sending? I have the information how it reports I can add support for it.

qba667, 2018-04-16:
@schwabe, sure I don't know if multi-protocol module uses and supports same format as I6/I6X but here you have list of sensors supported in cleanflight/betaflight:
https://github.com/qba667/FlySkyI6/blob/master/source/source/ibustelemetry.h

They are not official but we are using it for some time.

The variable length telemetry sensor packet structure is following:
0xAC (byte) TX_ID (4 bytes),  RX_ID (4 bytes) SENSOR_ID (byte) SENSOR_INDEX (byte) PAYLOAD_LENGTH (byte) PAYLOAD (n bytes) ........... 
frame may contains number of sensors but total length can not exceed 28 bytes. For example sending  BAAD F00D.
[image]
Standard implementations never sends sensors longer than 4 bytes, but we can pack sensors together and send them as single frame to deal with the limitation of IBUS sensors ids.

Backing to CAT-01 it sends 4 bytes payload with ID 0x41, the value contains temperature and pressure.
To get temperature you have to shift payload by 19 places to right, it is standard IBUS temperature (we have to subtract 40 to get real temperature). The pressure is expressed in pascals as 19 bit long value.

I am working on port to Cortex M0 (FlySky I6X) so maybe I can add complete telemetry handling for FlySky.

pascallanger, 2019-05-10:
Multi v1.2.1.44 is now forwarding the extended telemetry along with normal telemetry.
Can you look at the implementing a wrapper? (example of wrapper: https://github.com/qba667/FlySkyI6/blob/master/source/source/alt.c)

closed 2020-02-04 by stale[bot]