opentx/opentx #4958: Check for external antenna enabled when powering up or changing models

jeppling, 2017-05-27:
With 2.2.0N371 on Horus, I have a model is set to use internal+external antenna, When I power on the Horus with that model chosen, it has the external antenna enabled. While I have not yet powered on or chosen that model without the the external antenna chosen, I'm afraid that one of these days, that is going to happen. My understanding is that powering on a TX RF unit without an antenna attached will harm the RF unit.

It would be desirable for the software to check for enabled external antenna when powering up or when a model is chosen so that the software an pause and warn the user to check for external antenna or give the user an option to disable the external antenna before continuing. The user would have to explicitly select option to continue if an external antenna is attached. Maybe it should be an option to pause for external antenna check, so that those who don't want to be bothered with the warning can disable it.

I've so far never enabled a tranmit RF unit without the antenna, but I would hate to ruin the RF unit because I accidentally chose a model that uses an external antenna without the external antenna attached.

scott-adamson1975, 2019-02-12:
for x-lite too please

what would be perfect is that when turning on with a model  that has external antenna enabled when going thought the switch tests and throttle test also check if external antenna is enabled, warn, do not transmit until accepted, probably also give option to temporarily disable external antenna for programming, sd access etc. etc.

scott-adamson1975, 2019-02-12:
or... maybe better add some kind of lua script hook at on model selected, before transmit begin

3djc, 2019-02-12:
We have made changes to the model notes display so they are displayed while your module is off. Simply create a model note sayinbg : check external antenna or whatever else you feel like

scott-adamson1975, 2019-02-12:
ah, that would be edit checklist?

3djc, 2019-02-12:
Yes, you can do that from companion using edit checklist, manually using any text editor and saving the file at the right place, several options, as usual :)

scott-adamson1975, 2019-02-12:
would i be right in assuming that it doesn't start to transmit until you have pressed exit (in the case of the x-light) when viewing the checklist?

tried it with the simulator and it seams to do the job

3djc, 2019-02-12:
yes, as explained just above

scott-adamson1975, 2019-02-12:
perfect, many thanks

scott-adamson1975, 2019-02-15:
tried on latest firmware and checklist does not appear on Taranis Xlite, 

have enabled checklist in companion and entered a message but the message is not shown on powerup or on change of model, simulator appears different to actual

scott-adamson1975, 2019-02-15:
plus exit does not cancel selection of the model or turn off, by definition if i dont satisfy the checklist (i.e. push exist instead of ok) it shoudlnt continue to change model or power on

scott-adamson1975, 2019-02-15:
but most importantly ona physical taranis xlite with 2.2.3 firmware (eu, no heli, lua) teh checklist is not displayed

3djc, 2019-02-15:
Have you put the file on SDcard ?

scott-adamson1975, 2019-02-15:
i used the companion application and the option write models to radio after enabling checklist and modifying the text, does this not create the text file(s) in its correct place(s)?

3djc, 2019-02-15:
Doing that, Companion has created a checklist file on your PC sdcard path. If you want this file on your SD, you have to put it there yourself.

scott-adamson1975, 2019-02-15:
ah, i thought it was included in the write models to radio routine of the companion

3djc, 2019-02-15:
No, that routine does not touch SD card content (on Taranis, on Horus it just write radio and models settings, not anything else either)

scott-adamson1975, 2019-02-15:
i see now there is an option in the file menu, synchronise SD, i guess that will copy it over, will try later

bsongis, 2019-09-05:
Done

closed 2019-09-10 by 3djc