opentx/opentx #4976: After upgrade to nightly build i can't use QX7 for simulators.

crocodanser, 2017-05-30:
After upgrade to nightly build i can't use my Taranis QX7 for simulators.
There is just Frsky Radio USB device in mass storage device in the device manager of windows 10.
In stock firmware i don't have this issue.
I tried RC15 and the lastest and i have the same issue after upgrade.
When i launch liftoff or other simulator i can't  configure my controller.
I need use nightly build for D8 mod.
If i roll back to manufactory firmware, the problem disapear.

Any idea?

Thx in advance
Sorry for my english

schwabe, 2017-05-30:
do not use the option massstorage when installing firmware from companion

closed 2017-05-30 by schwabe

crocodanser, 2017-05-30:
ok i will try that . Thx