opentx/opentx #4978: False Model ID already used

airhamer, 2017-05-31:
Did a search, and saw some old closed ones, but not a match, and none open.  Opening this in an attempt to help with testing and information.

Taranis X9D plus

Updated to new 2.2.0 firmware.
SDCard updated.

How to Reproduce:
Cleaned off all model data
Calibrated etc....
Create second model, and turned off both internal and external radios for use in simulator via USB.
Delete Model01 default model.
Move "Simulator" model to position 1.
Now try to create a model in position 2.  The "Model ID already used" warning shows up.
I did it this way after I had problems with a full suite of models and moving things around and was unable to show a friend how to use the new model wizard.  So.. I cleaned everything off to isolate the problem as best I could.

Three problems with this...  
First, it isn't warning me that I would bind to a receiver (bind) number that might be used which is why I think this might be useful.
Second, It is incorrect in that no models are using any receiver bind numbers.
Third, there is no way to acknowledge and continue to the "New Model Wizard".

kilrah, 2017-05-31:
Cannot reproduce with your info, there is no warning when creating the 2nd model. Your "problems" are also contradictory, you're saying that it isn't warning you, yet you're saying it's showing the warning... makes no sense.

airhamer, 2017-06-05:
Kilrah,
I have seen many posts of yours that have helped me understand things better in OpenTX and other RC related things., and I respect you.

I am surprised that you cannot replicate it.  I did the best I could think to isolate it.   Totally cleaned off the radio etc.  I will blow all my settings away again.. If it still happens, would a video be helpful?

Let me explain what was contradictory...

I get a warning that says "model ID already used"  however it is in fact not used. because the only other model that exists in the memory does not have either external or internal radio on, and has never been bound to a receiver.  At the moment the error occurrs, there is only one model in memory, and it is in position one with no radio turned on.  ????  Video needed?

I can see why it might get that idea though, because the model in memory position one on the list was created in memory position two.  So it doesn't seem to key off of the receiver number in the bind section of the model memory???

If, however, this message did come up and model one was bound and used a receiver ID of 2, I would expect to be able to acknowledge the message and continue to the wizard.

3djc, 2017-06-06:
Yes, a video of the whole sequence starting with blank radio would help, as we cannot reproduce

jtaylor2, 2017-06-06:
I can recreate his problem in the x9d 2.2.0 simulator.  Steps to recreate:
1. Create model 1  - Radio mode D16 Receiver No. 2
2. Turn off internal receiver (both RX are off)
3. Go to model 02 and create a model - Message box "model ID already used" appears
4. Hit exit to dismiss message
5. Default model is created without going to the Wizard

Johnex, 2017-06-06:
I saw something similar that might be related.
When i made a few models and i deleted some, i wanted to change the reciever id's to be sequential. During that process there was a model with ID 02 for example and i changed it to 10, and when trying to change another model to ID 02, which is no longer used cos i changed it, it still said that the ID was being used.

3djc, 2017-06-06:
I think it opens an interesting question : what 'frees' an ID, and what does not ? Is turning off a module a real sign that this ID is not used anymore ? Maybe not so, you may simply be working on a complex model, and you disabled radio not be bothered and save battery life.

schwabe, 2017-06-06:
A real fix probably has to wait until we do an EEPROM change. At the moment also a DSM model with model id 2 conflicts with a XJT model with model id 2. But we don't have that information make any better decision available.

powercycle, 2017-06-06:
I can't reproduce this if I use companion only to create the models.
However, I can reproduce this using the simulator.
If I create the models in companion with the same ID, it doesn't complain when I use either model.  
If I change an existing model ID to something that already exists on the simulator, it gives me the warning.

Johnex, 2017-06-06:
I only tested it in the actual radio, so simulator would give the same results.

jtaylor2, 2017-06-06:
Leaving the question of if turning off RF frees the receiver id open for debate, I think adding to the problem is that the id in use message doesn't tell you which model is using it and if you have the RF turned off the id is not even displayed so you can't find it.  Second it saya "Model ID" and not "Receiver Number" so they don't really know where to look.  And clearing the message creates the model without giving the opportunity to use the Wizard.

kilrah, 2017-06-07:
In the simu for me the wizard runs after pressing exit to dismiss the warning. You can run it manually anytime you want anyway.

This behavior is normal with the current implementation. As posted earlier there is already an open discussion about improving it, so please continue in #2187.

closed 2017-06-07 by kilrah