opentx/opentx #4989: X7 SD Card Contents S6 Calibration Images For Reference

RCdiy, 2017-06-01:
It would be beneficial to add these images to the S6 folder or sub folder.
1 Label Up.png
[image]
2 Label Down.png
[image]
3 Pins Up.png
[image]
4 Pins Down.png
[image]
5 Label Facing You Pins Right.png
[image]
6 Label Facing You Pins Left.png
[image]

[Archive.zip](https://github.com/opentx/opentx/files/1045058/Archive.zip)

schwabe, 2017-06-01:
Could you eloborate? We already have pictures in X9 and Horus. For the X7 the images were to difficult to view on the remote iirc. @3djc can probably eloborate why we don't have pictures there.

3djc, 2017-06-01:
Given the very limited screen capability of x7, we do not handle pictures from Lua. They might seem to look OK on a pc screen, they would less so on the real lcd screen. That's why we went to the least ambiguous text we could think of

closed 2017-06-01 by 3djc