opentx/opentx #4990: QX7 2.2 release Editing errors on the radio

rdeanchurch, 2017-06-01:
I have a model with Mixes on the upper channels.  On the radio I tried to edit them to change elevator from + to - and visa versa on ch 10 and 11, It is a wing.  Extra mixes using Trim as a source were generated/inserted on ch11 and 12.   I use TREA channel order.
(This is for a static demo of the QX7 controlling 2 models to provide some motion instead of just a static demo.  Not for flight.  A conventional 4 channel and a wing.)

Ignore Ch04 error with offset 50, My goof.

[image]

kilrah, 2017-06-01:
Is that visible ont he radio too, or only when importing into companion?

rdeanchurch, 2017-06-01:
Not visible on the radio.  Only when imported to Companion.  If fact visibility of of multiple mixes is missing on the radio completely on Ch9 thru Ch12..(all I have tested.)
This was repeatable, after I saw it the first time, I thought I had fumble fingered something.  I re loaded the radio from companion and it was repeatable.

kilrah, 2017-06-01:
If it's repeatable, can you give the steps to reproduce?

rdeanchurch, 2017-06-01:
1. I need to explore further.  I may have caused this with my Lua script to move surfaces for my demo.
I'll report back after exploring further.
2.  Yes, surfaces all move correctly, so mixes are there.  But when I look at channel 9 thru channel 11 only one mix is showing when look at the model via menu-short, page over to mixes and look  at channels  9 thru 12
Channel 9 and 10 show nothing and Channel 11 and 12 show the first mix of channel 10 and 11.

But I need to disable my lua script and test again.

rdeanchurch, 2017-06-02:
My lua script was apparently causing this.  Closed by originator. 
Sorry for the false alarm.

closed 2017-06-02 by rdeanchurch