opentx/opentx #4992: Companion: RS & LS Inputs Missing

AustinSaintAubin, 2017-06-02:
Right & Left Sliders (RS & LS) inputs missing from Companion 2.2.0 for Taranis Plus (any maybe others). RS & LS can be added as inputs from the radio interface, but not through companion.

[image]

[image]

powercycle, 2017-06-02:
Check in the hardware settings to ensure they are enabled.

AustinSaintAubin, 2017-06-02:
You are correct Sir. The Hardware LS & RS was set to none. Which is odd because on the radio I was still able to assign RS & LS for use. Not sure how this was disabled. Thanks for your help.
 
[image]

closed 2017-06-02 by AustinSaintAubin

RaccoonEye, 2017-08-04:
Where is the Hardware menu ? I can't find it. Thanks.

powercycle, 2017-08-05:
[image]

pghern, 2018-01-04:
I have changed these settings to access LS/RS in Companion but the settings do not save.... each time I start Companion I have to reset LS/RS in the Radio Settings - Painfull.