opentx/opentx #4996: [Companion] Export-import companion configuration

OpenMik, 2017-06-03:
Companion has different configurations: radio profile, simulator settings, application settings etc.
May be useful to provide a function that allow export and import all settings in a file,  in case of use on different PC or system reset a new installation may be simplified

powercycle, 2017-06-03:
+1.  It would be nice to load the OTX file and have the radio profile that it was created under instead of warning you it has been converted.

kilrah, 2017-06-03:
@powercycle that's not at all what this is about.

powercycle, 2017-06-04:
Should have worded that better, but I believe that if you could do this, then my additional item would be met as well.

mpaperno, 2017-06-04:
👍  on config export/import.  Shouldn't be hard to implement, mostly some UI work to provide the functions.  (We could use INI files as that is already a native format for QSettings.)

Perhaps the ability to export/import individual radio profiles could also be useful?  Maybe if just for troubleshooting... make it easier to exchange settings.  Considering that there's also some radio data possibly stored in the profiles as well (not sure how many people use that).

@powercycle Is that what you're referring to?  It sounds like you want to auto-select an existing profile which matches whichever file you're opening.  In which case I don't see how your suggestion relates to this one. You should start a new issue for that if you wish to discuss it.  I think it's a valid idea but there are some (possibly major) hurdles to overcome in order to pick the correct profile based on data in a radio file... in some cases it may not even be possible to determine the exact radio type at all, or which firmware options were selected.

powercycle, 2017-06-05:
I'll open another issue.

OpenMik, 2017-06-05:
my request is that you can export the Companion configuration in a single file as if it were a snapshot so that you may restore it as before in the case of multiple installations or system recovery
This includes all settings and radio profiles that are not individually saved in this case.

PR #6006 (2018-07-05): Refactor Companion settings system
 + All stored properties are fully dynamic, using an extended version of the Qt Property System;
 + CompStoreObj is now QObject subclass;
 + All properties now have change notifier signals and reset (to default) handlers;
 + Do not store default values in settings;
 + Settings import now ignores all non-existent properties automatically;
 + Add settings export functions;
 + Add GUI and CLI methods for settings export and import to/from file (closes #4996);
 + On first start, offer option to import settings from file (as well as previous version if found);
 +  [... cut]
closed 2019-11-15 by stale[bot]