opentx/opentx #4999: [HORUS] Crossfire Lua Script

mdrjr, 2017-06-04:
Is it possible to run the Taranis Crossfire lua on the Horus? 
Just killing the lcd.drawScreenTitle() makes it work?

3djc, 2017-06-04:
There is no technical reason why it can't be run on Horus, it is just that no one in the team does have the required equipment to test and debug it on Horus.

Some work is surely needed.

mdrjr, 2017-06-04:
I've just ordered a Crossfire so I'll fix the script in the next few days. :) 
once I get it done I'll send a PR with it.

closed 2017-06-04 by mdrjr

IdefixRC, 2017-08-15:
Hi guys, is there any progress on this? I have played around with the script and updated it to show the devices as well as the first device screen but I'm stuck with the conversion of the old script around the pages and lcd.getLastPos() function which is not supported by Horus FW.
Thanks !

mdrjr, 2017-08-15:
@IdefixRC I have no interest in the Horus anymore, it doesn't fit my needs.

SoulPictures, 2017-08-22:
Yesterday I just played around with the script, sadly I have not yet decided whether to get CF or not but as far as I got the stuff tested I got the script running on my horus

IdefixRC, 2017-08-22:
Continued [https://github.com/opentx/opentx/issues/5162](url)