opentx/opentx #5000: S6R LUA Script "hangs" on page 2 whilst loading settings

jasonhyland, 2017-06-04:
I am using the latest (2.2.0 31st May) LUA script to confirgure the S6R from my Taranis+. It appes to hang when loading from page 2. I can connect and configure from the STK.

Please see this video for an example;

https://youtu.be/nxSuvrpn7ZU

Thanks, Jason

3djc, 2017-06-04:
Please make sure your s6r is on the last firmware

jasonhyland, 2017-06-04:
Hi, it was flashed with version 161129 which I understand is the latest? Interstingly if you look at the FrsSky page [here](http://www.frsky-rc.com/download/view.php?sort=Firmware&down=285&file=Firmware-%20S6R) it's offering version 160909 on the download link. I've tried with both versions.

jasonhyland, 2017-06-04:
I'll try the BETA firmware (170217), perhaps this is what you mean?

jasonhyland, 2017-06-04:
OK, works with the BETA 170217 firmare. Thanks.

closed 2017-06-04 by jasonhyland