opentx/opentx #5003: Feature Request: ability to add additional telemetry entries/names and parameters in telemetry simulator.

jsonyu, 2017-06-05:
The telemetry simulator contains a fixed set of telemetry items. It would be great if there's a way for to add several custom telemetry names/items in there.  

I understand it cant be unlimited but maybe at 5-10 empty entries where one can enter a telemetry name, id, value, offset etc would be a great feature and make it easier to test custom scripts.

Many thanks in advance!

mpaperno, 2017-06-05:
Can't promise anything, but I *may* have something like that working soon.  My idea is that you'd be able to configure any number of sensors (pre-set types or completely custom) and/or read them from the current model.  Sensor "sets" could then be saved/loaded, perhaps also including the current values.  This would also allow log files with custom/unknown/translated sensor names to be loaded/played.

Related: #4256, #4434

jsonyu, 2017-06-05:
@mpaperno that's exactly what I mean.. this is great! looking forward to this feature. Many thanks!

closed 2019-11-15 by stale[bot]