opentx/opentx #5007: Very long startup time of QX7 after update to official 2.2.0

franz4949, 2017-06-05:
After update of my QX7 to official release 2.2.0 it takes about 3 seconds for startup. During that startup time the display shows "OPEN TX". In the Tx setup page I have not activated the option to show a startup picture. Can change to any model, also without SD card no change in loading time. When pressing a key e.g. ENTER it immediately skips/terminates the splash screen. My last OTx version before this 2.2.0 was the nightly N368, startup time there was much shorter.

franz4949, 2017-06-06:
Excellent, thank you. Does #5008 mean that it's now implemented and I will have the modification available with a new download of 2.2.0 ? And shall I close this open issue?

3djc, 2017-06-06:
It will be available in 2.2.1 when that gets released, and the issue will close itself after code review

franz4949, 2017-06-06:
Okay, thanks again.

PR #5008 (2017-06-28): Fully adjustable splash timer on x7, matching x9 implementation
[image]

This fixes #5007
closed 2017-06-28 by bsongis