opentx/opentx #5013: Betaflight Lua Scripts not working on newest OpenTx version

Tiim12312, 2017-06-06:
I just updated everything on my QX7 to use the PID changing Lua scripts on it. The Lua scripts come up with nothing in the fields for PIDS and Rates and VTX. Even stranger is that when I have the script running then plug in the quad whatever Lua menu I had up will populate but will not allow me to save changes and if I switch to a different menu say rates of vtx those are blank. Please help!

schwabe, 2017-06-06:
Check if your normal telemetry is coming regulary in the telemetry screen. I had update my XSR to its beta firmware to get BF telemetry working again.

Tiim12312, 2017-06-06:
I am getting RSSI and voltage but nothing else. Is that ok?

schwabe, 2017-06-06:
No, then you getting no telemetry at all from Betaflight. Try discover telemetry on and power cycle your quad. If you have the BF sensors for a few seconds and then it stops you have the same as I which I could only solve by updating my XSR. Checking with my osciloscope even showed that the XSR was querying for telemetry and BF answered but somehowed it did not make it back to the other side.

Btw. this a issue that only Frsky knows what is going on. 

Btw. did you also update your internal XJT firmware or is that unchanged in your case?

Tiim12312, 2017-06-06:
I did not update either of those and am trying now. Thank you!

3djc, 2017-06-07:
Not an opentx issue by the look of it, closing

closed 2017-06-07 by 3djc