opentx/opentx #5016: Companion 2.2 crashes on startup MacOS and Windows 10

cobez83, 2017-06-07:
Just downloaded the latest stable release of OpenTX Companion v2.2 and found that it crashes every time on start up on both MacOS 10.12.3 as well as Windows 10 (with anniversary edition update), currently using Companion v2.0.17 with no issues.

See below dump of MacOS error report when launching, thanks in advance...

---

Process:               companion [36449]
Path:                  /Volumes/VOLUME/*/OpenTX Companion 2.2.app/Contents/MacOS/companion
Identifier:            ???
Version:               ??? (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           companion [36449]
User ID:               501

Date/Time:             2017-06-07 21:58:13.316 +1000
OS Version:            Mac OS X 10.12.3 (16D32)
Report Version:        12
Anonymous UUID:        537ECD08-9F28-77CB-BEF5-BCAF58BC2DE5

Sleep/Wake UUID:       6020FB39-49D5-46AF-849D-BE710948C744

Time Awake Since Boot: 150000 seconds
Time Since Wake:       1700 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.qt-project.QtCore         	0x000000010afd4109 QSettings::allKeys() const + 25
1                                 	0x000000010965b1ff AppData::importSettings(QSettings&) + 1695
2                                 	0x0000000109657ba5 AppData::init() + 277
3                                 	0x0000000109385cf4 main + 420
4   libdyld.dylib                 	0x00007fff99524255 start + 1

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff996534e2 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff9973b791 _pthread_wqthread + 1426
2   libsystem_pthread.dylib       	0x00007fff9973b1ed start_wqthread + 13

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff996534e2 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff9973b791 _pthread_wqthread + 1426
2   libsystem_pthread.dylib       	0x00007fff9973b1ed start_wqthread + 13

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff996534e2 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff9973b5fe _pthread_wqthread + 1023
2   libsystem_pthread.dylib       	0x00007fff9973b1ed start_wqthread + 13

Thread 4:
0   libsystem_kernel.dylib        	0x00007fff996534e2 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff9973b791 _pthread_wqthread + 1426
2   libsystem_pthread.dylib       	0x00007fff9973b1ed start_wqthread + 13

Thread 5:
0   libsystem_kernel.dylib        	0x00007fff996534e2 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff9973b791 _pthread_wqthread + 1426
2   libsystem_pthread.dylib       	0x00007fff9973b1ed start_wqthread + 13

Thread 6:
[... 277 lines cut ...]
VM_ALLOCATE                         60K        9 
__DATA                            21.7M      226 
__GLSLBUILTINS                    2588K        2 
__IMAGE                            528K        2 
__LINKEDIT                       118.5M       17 
__TEXT                           163.0M      224 
__UNICODE                          556K        2 
mapped file                       45.9M       11 
shared memory                     16.4M       15 
===========                     =======  ======= 
TOTAL                            521.8M      556 

Model: MacBookPro11,2, BootROM MBP112.0138.B21, 4 processors, Intel Core i7, 2.3 GHz, 8 GB, SMC 2.18f15
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343531533641465238412D50422020
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343531533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (7.21.171.68.1a5)
Bluetooth: Version 5.0.3f1, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0512F, 500.28 GB
USB Device: USB 3.0 Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1

3djc, 2017-06-07:
Clearly, if that was the case, we would have had a LOT of reports already.

Are you working or doings things QT related ?

cobez83, 2017-06-07:
Understandably but can be replicated on 2 separate windows 10 boxes and on MacOS, this issue is on launch of the application, I don't even get to the splash screen.

3djc, 2017-06-07:
And I understand that, but if out of several tenth of thousands users, you are the only one having this issue, there must be something very unique on your side that we cannot really guess alone

schwabe, 2017-06-07:
Can you to delte all ```~/Library/Preferences/com.opentx.Companion 2.2.plist`` and similar files?

cobez83, 2017-06-07:
Checked the above location on the Mac and no mention of com.opentx.xxxx files.

schwabe, 2017-06-07:
A crash in that code path can only happen if OpenTX decides to import settings from a previous version. And also since you have Companion 2.0.x running, there should be a preferences file

cobez83, 2017-06-07:
I'll have a look and see if i can find anything relating on the drive to this. Makes since that the file should exist somewhere.

cobez83, 2017-06-07:
The only files that contain "opentx" are the downloaded dmg and the the .app file, this is strange...
The only common on all PCs that I can think of is they all at one time have had companion installed.

cobez83, 2017-06-07:
OK, progress. Thanks schwabe, on one of the windows boxes i delete all references to companion and opentx which fixed the issue.

On the MacOS I deleted the same referenced names .plist and it worked... Looks like the previous version used is causing some sort of conflict with the newer version v2.2.

All good now and thank you all for the help.

closed 2017-06-07 by cobez83

mpaperno, 2017-06-07:
This was, in fact, a bug in Companion introduced after RC18 (since 0fc1681, fixed with 655e2f9).

lennelei, 2017-06-18:
Hi, just for information (in case someone else get there) : I had the same problem on Windows 10 (crash of OpenTX 2.2.0 in Qt5Core.dll) and I solved it by removing all old registry information under the keys HKEY_CURRENT_USER\SOFTWARE\OpenTX\Companion 2.0 and HKEY_CURRENT_USER\SOFTWARE\OpenTX\Companion 2.1 (I found no files to remove).

rhyde, 2017-08-12:
I am seeing a similar crash on 2.2 which I download yesterday.  Deleting the previous plist for 2.0 did not fix it.

Process:               companion [15280]
Path:                  /Applications/OpenTX Companion 2.2.app/Contents/MacOS/companion
Identifier:            ???
Version:               ??? (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           companion [15280]
User ID:               501

Date/Time:             2017-08-11 18:25:07.241 -0700
OS Version:            Mac OS X 10.10.5 (14F2511)
Report Version:        11
Anonymous UUID:        6F29948F-76C4-6A12-97D8-A249FB8C7EE7

Time Awake Since Boot: 70000 seconds

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000

VM Regions Near 0:
--> 
    __TEXT                 000000010b614000-000000010c061000 [ 10.3M] r-x/rwx SM=COW  /Applications/OpenTX Companion 2.2.app/Contents/MacOS/companion

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.qt-project.QtCore         	0x000000010d57f3e0 typeinfo name for QObjectData + 0
1                                 	0x000000010b9c91ff AppData::importSettings(QSettings&) + 1695
2                                 	0x000000010b9c5ba5 AppData::init() + 277
3                                 	0x000000010b6f3cf4 main + 420
4   libdyld.dylib                 	0x00007fff8ec295c9 start + 1

Thread 1:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff8e627232 kevent64 + 10
1   libdispatch.dylib             	0x00007fff84b1b26e _dispatch_mgr_thread + 52

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff8e62694a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8f2763dd start_wqthread + 13

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff8e62694a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8f2763dd start_wqthread + 13

Thread 4:
0   libsystem_kernel.dylib        	0x00007fff8e62694a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8f2763dd start_wqthread + 13

Thread 5:
0   libsystem_kernel.dylib        	0x00007fff8e62694a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8f2763dd start_wqthread + 13

Thread 6:
0   libsystem_kernel.dylib        	0x00007fff8e62694a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8f2763dd start_wqthread + 13

Thread 7:
0   libsystem_kernel.dylib        	0x00007fff8e62694a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8f2763dd start_wqthread + 13

Thread 8:
0   libsystem_kernel.dylib        	0x00007fff8e62694a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8f2763dd start_wqthread + 13

Thread 9:
0   libsystem_kernel.dylib        	0x00007fff8e62694a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8f2763dd start_wqthread + 13

[... 275 lines cut ...]
TOTAL                             485.8M
TOTAL, minus reserved VM space    485.3M
 

Model: iMac15,1, BootROM IM151.0207.B08, 4 processors, Intel Core i7, 4 GHz, 16 GB, SMC 2.23f11
Graphics: AMD Radeon R9 M295X, AMD Radeon R9 M295X, PCIe, 4096 MB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x802C, 0x31364B544631473634485A2D314736453220
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x802C, 0x31364B544631473634485A2D314736453220
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x142), Broadcom BCM43xx 1.0 (7.15.166.24.3)
Bluetooth: Version 4.3.6f4 17916, 3 services, 27 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Serial ATA Device: APPLE SSD SM0512F, 500.28 GB
USB Device: USB3.0 Hub
USB Device: USB3.0 Hub
USB Device: USB3.0 Hub
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB2.0 Hub
USB Device: USB2.0 Hub
USB Device: USB2.0 Hub
USB Device: Microsoft® Nano Transceiver v2.1
USB Device: Corsair Vengeance M60 Mouse
USB Device: Apple USB SuperDrive
Thunderbolt Bus: iMac, Apple Inc., 26.1

fiveangle, 2017-08-14:
Same here…

<details><summary>Mac OS crash log...</summary>

Process:               companion [14262]
Path:                  /Applications/OpenTX Companion 2.2.app/Contents/MacOS/companion
Identifier:            ???
Version:               ??? (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           companion [14262]
User ID:               504

Date/Time:             2017-08-14 00:43:31.308 -0700
OS Version:            Mac OS X 10.12.6 (16G29)
Report Version:        12
Anonymous UUID:        AFFE86A5-86EC-526B-638D-922D056ABE00

Sleep/Wake UUID:       6F8C89E9-BAE8-4D24-9692-A450A1F3B6F5

Time Awake Since Boot: 47000 seconds
Time Since Wake:       9400 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [0]

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.qt-project.QtCore         	0x000000010b9c8109 QSettings::allKeys() const + 25
1                                 	0x000000010a05c1ff AppData::importSettings(QSettings&) + 1695
2                                 	0x000000010a058ba5 AppData::init() + 277
3                                 	0x0000000109d86cf4 main + 420
4   libdyld.dylib                 	0x000000010be9c235 start + 1

Thread 1:
0   libsystem_kernel.dylib        	0x000000010befa44e __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x000000010bfd4621 _pthread_wqthread + 1426
2   libsystem_pthread.dylib       	0x000000010bfd407d start_wqthread + 13

Thread 2:
0   libsystem_kernel.dylib        	0x000000010befa44e __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x000000010bfd4621 _pthread_wqthread + 1426
2   libsystem_pthread.dylib       	0x000000010bfd407d start_wqthread + 13

Thread 3:
0   libsystem_kernel.dylib        	0x000000010befa44e __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x000000010bfd4621 _pthread_wqthread + 1426
2   libsystem_pthread.dylib       	0x000000010bfd407d start_wqthread + 13

Thread 4:
0   libsystem_kernel.dylib        	0x000000010befa44e __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x000000010bfd4621 _pthread_wqthread + 1426
2   libsystem_pthread.dylib       	0x000000010bfd407d start_wqthread + 13

Thread 5:
0   libsystem_kernel.dylib        	0x000000010befa44e __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x000000010bfd4621 _pthread_wqthread + 1426
2   libsystem_pthread.dylib       	0x000000010bfd407d start_wqthread + 13

Thread 6:
0   libsystem_kernel.dylib        	0x000000010befa44e __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x000000010bfd4621 _pthread_wqthread + 1426
[... 286 lines cut ...]
shared memory                     16.8M       13 
===========                     =======  ======= 
TOTAL                            566.2M      600 

Model: MacBookPro11,3, BootROM MBP112.0138.B40, 4 processors, Intel Core i7, 2.6 GHz, 16 GB, SMC 2.19f12
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
Graphics: NVIDIA GeForce GT 750M, NVIDIA GeForce GT 750M, PCIe, 2048 MB
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (7.21.171.130.1a1)
Bluetooth: Version 5.0.5f1, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM1024G, 1 TB
USB Device: USB 3.0 Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1
</details>

Deleting `~/Library/Preferences/com.opentx.OpenTX Companion.plist` allows to launch.

plist that Companion 2.2 is choking on:

[com.opentx.OpenTX Companion.plist-bad.zip](https://github.com/opentx/opentx/files/1221585/com.opentx.OpenTX.Companion.plist-bad.zip)

commit 655e2f90 (2017-06-07): [Companion][AppData] Fix possible crash when importing settings from Companion < v2.1 (fixes #5016).