opentx/opentx #5017: OpenTX 2.2.0 RC7 (N359): Aerofly RC 7 does not correctly read channel values

zkhalapyan, 2017-06-08:
Upgraded to the latest version from OpenTX 2.1, and after the upgrade, Aerofly RC 7 doesn't recognize the channel values. 

[image]
[image]

3djc, 2017-06-08:
Why are you updating to a several month old beta version when 2.2.0 is available : http://www.open-tx.org/2017/05/30/opentx-2.2.0 ?

projectkk2glider, 2017-06-08:
He is not using OpenTX RC7, but Aerofly RC 7.

@zkhalapyan do the channels move on the radio LCD screen when you move the sticks? Looks like the USB joystick is recognized.

3djc, 2017-06-08:
Well it says also n359...

kilrah, 2017-06-08:
> He is not using OpenTX RC7, but Aerofly RC 7. 

In the title...

Westone64, 2018-05-20:
zkhalapyan, was this issue able to be resolved? I am experiencing exactly this issue with a Taranis Q X7S running the stock openTX-x7s 2.2.0, with AeroFly RC7 on a Mac running High Sierra. So far unable to find a fix.

Thanks.

closed 2019-11-15 by stale[bot]