opentx/opentx #5019: Companion 2.2.0 telemetry 'switches' not available in custom functions

3djc, 2017-06-09:
On the radio, you can setup special function based on a sensor been true or not

You cannot do that on companion, and if you import a radio with such a special function, the line stay empty

[image]

And yes, we do have a little issue on the screen, I'll deal with that ;)

mpaperno, 2017-06-09:
Also the `Telem[etry streaming]` "switch" is missing, right?  We don't even have enums for either of those, so it's no wonder they're missing in the lists. :)

Are there any screens/contexts besides the Special Functions where sensor "switches" should be shown?  (They're already shown in `RawSource` selectors in some cases.)

```cpp
  enum SwitchContext
  {
    LogicalSwitchesContext,
    SpecialFunctionsContext,
    GlobalFunctionsContext,
    TimersContext,
    MixesContext
  };
```

> ... based on a sensor been true or not

Meaning `true` if the sensor value != 0, or based on sensor being present/active?

kilrah, 2017-06-09:
> Also the Telem[etry streaming] "switch" is missing, right? 

Nope this one is there

[image]

mpaperno, 2017-06-09:
I meant in Companion.

kilrah, 2017-06-09:
Ooh yes... too early :\

tavdog, 2017-11-26:
This issue seems it should be an easy fix but I can't find anywhere in the source where all the options are listed.

kilrah, 2017-11-26:
That was fixed long ago.

closed 2017-11-26 by kilrah

tavdog, 2017-11-26:
This bug is still present in companion 2.2.  When setting up Telem switch in the transmitter, then opening the config in companion, the Telem switch goes to --.