opentx/opentx #5021: Telemetry loop taranis QX7 CRSF protocol

stevef1966, 2017-06-10:
Hi seem to have an issue using CRSF on a taranis QX7 with the TBS crossfire, all devices on the latest recommend firmware, CRSF works but the QX7 keeps reporting telemetry lost/recovered.

If I switch top PPM to talk to the crossfire the warnings from the qx7 stop

Have checked the pins in the module bay and checked the soldered connection of the pins to the qx7
Have even jumpered the pins from the qx7 to the crossfire thx but still the same

3djc, 2017-06-10:
Duplicate #4922

closed 2017-06-10 by 3djc

stevef1966, 2017-06-10:
apologies, didn't realise it had been included as an issue, thought id have to open a new issue, will know for next time - keep up the great work