opentx/opentx #5024: Center Beep For the Right Slider5 RS

ictea-rc, 2017-06-11:
In the Setup Page of the model if checking the box for the Center Beep for the Right Slider RS only, instead of playing the "beep" track when RS is centered, the following is played: "Timer 1 elapsed".  The LS box is working fine.

I think it's a firmware issue, because it happens in the Companion as well as programing the radio directly.

powercycle, 2017-06-11:
I can't reproduce this on the transmitter (X9D+) using May-31-2017 2.2 release.  
You don't mention radio type.

powercycle, 2017-06-11:
I tried to use the simulator for the X7 and Horus, but don't know how to make the item go up and down (like +\- on X9D+)

ictea-rc, 2017-06-12:
The radio is X9E w/ 2.2.0

3djc, 2017-06-12:
This is a bug indeed

PR #5028 (2017-06-14): Adjust audio for X9E hardware
Because world best radio has four sliders by default ;)

This fixes #5024
closed 2017-06-14 by bsongis