opentx/opentx #5029: feature request: add "copy & paste" functionality for complete mixer channels

strgaltdel, 2017-06-12:
Hi,

after intensive rearrangement of mixer-line configuration,
i found that an additional "copy & paste" functionality  regarding a complete mixer channel, not only the lines, would be helpful.

Even when you try to follow own "standard operation procedures" including fix channel/mixer line sequences in templates, sometimes you'll run into situations where you have to move complete "mixer channels"
i.e. imagine you you're exchanging receiver types in planes from X8r to S8R, or configuring  6- and 8-flap glider based on a 4-Flap template.
Usually one flap-channel is defined by 4..6 Mixer lines, which generates excessive "cut&paste" operations in an 6 or 8-flap environment with a risk of moving a line into the wrong channel.

[image]

By migrating 3 planes in the last week, i thought about the comfort of such a functionality several times,
maybe it could find a way for implementation in one of the future releases.

Thanks and regards

Udo

btw
copy/paste functionality in terms of SF & LS seems to be a little bit "hidden" ?
maybe it could be more intuitive for users if a frame like in the "#" or "switch" header would surround the LS / SF descriptors?

[image]

closed 2018-06-05 by mpaperno

mpaperno, 2018-06-05:
This is now possible by way of multi-selecting all the mixes on one channel (with mouse and CTRL/SHIFT key) and then dragging them all at once to a new channel (remember to hold down SHIFT to move vs. COPY), or using cut/copy and paste.

Same is true for the Inputs.

It also possible to copy/move any number of lines to a new/existing channel/input.

The only caveat is that you can't copy/move multiple *channels* at once... all the selected lines would end up in the same destination channel.

strgaltdel, 2018-06-05:
Ich kehre zurück am 11.06.2018.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "Re: [opentx/opentx] feature
request: add "copy & paste" functionality for complete mixer channels (#5029)" gesendet am
05.06.2018 09:45:17.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.

strgaltdel, 2018-06-05:
Ich kehre zurück am 11.06.2018.

Guten Tag,

ich bin bis zum o.g. Termin nicht erreichbar.
In dringenden Fällen können sie telefonisch die Zentrale unter 02173 950-5 erreichen und sich mit
dem jeweils zuständigen Kollegen verbinden lassen.

Danke

Hinweis: Dies ist eine automatische Antwort auf Ihre Nachricht  "Re: [opentx/opentx] feature
request: add "copy & paste" functionality for complete mixer channels (#5029)" gesendet am
05.06.2018 09:35:13.

Diese ist die einzige Benachrichtigung, die Sie empfangen werden, während diese Person abwesend ist.

commit 0a4ca595 (2018-06-05): [Companion][MixersListWidget] Enable multiple selection capability on Inputs and Mixes editors (closes #5029); Fix bug with custom style propagation; cosmetics.