opentx/opentx #5030: Enhancement new special function action copy trims to subtrims

elecpower, 2017-06-13:
The request for this action was mentioned in a Painless360 video on Plane Maiden - Tips and Tricks.
Easier to perform the action with a switch when trim adjustment limit reached in flight.

powercycle, 2017-06-13:
Try Instant Trim special function.  I think that will do what you want.

kilrah, 2017-06-13:
And much easier to make your model unflyable by a single button press too.
You can use extended trims if you expect such deviation.

closed 2017-06-13 by kilrah

elecpower, 2017-06-13:
Yes Instant Trim does to a degree but up until you run out of trim. Yes Extended Trim can further assist. Yes if you have an extra pair of hands on the flight line you can perform the copy.
> And much easier to make your model unflyable by a single button press too
There are many places not just special functions in a model config to make a model unflyable. The use of this requested action like any of those methods even if used erroneously should be up to the pilot. This is the beauty of OpenTX and why it is so popular. If you want out of box limited functionality you buy a name brand radio with proprietry software.
Please consider reopening.

kilrah, 2017-06-13:
Again Instant trim with Extended trims enabled does exactly that, so no.