opentx/opentx #5034: Turnigy 9X Stockboard (Atmega64) didn't convert the eeprom " Warnung" occurred

Wesie, 2017-06-16:
In the last week, I've changed the SW of 2 transmitter. One TX was with the factory FW from HK ( Turnigy) , the other SW was ... sorry .. I have forgotten the name.
Both boards contain the ATMEGA64A  ( 1 time Version 1 / 1 time Version2 Board ) 
I've selected the 9x Stock board and have download the OptenTX 2.2 370 SW.
After flashing a warning occurred in the Display and after keypress the Eeprom was initialised.
the first standard Model was visibel.
Uncomfortable, I received at every time, if I download or upload the modells with Companion an Eeprom Warning message, that the Eeprom is out-dated and I should save the file a backup and upgrade the Eeprom.
The eeprom Version is in both case Vers 216-1
I hvae search in the interenet, how I can upgrade ot DL the Version 218 ? so like the SD Card Change with my X9E; but I haven't found anything. 
Also a Backup and restore the Model-file via Companion doesn't help.
What does I've read in the menatime, that the Vers. 216 matched with Version 2.0 / 217 with 2.1 and 218 with 2.2 (?) .

I believe, that the conversion - also after a brandnew install doesn't worked as expected, or the Version No. is not updated.

projectkk2glider, 2017-06-24:
First of make sure you use the latest 2.2 official version, from your description it looks like you are using a nightly build 370. Then report back if issues persist.

powercycle, 2017-06-24:
It seems also possible that you have wrong SD data.

kilrah, 2017-06-24:
There is no SD on a 9x...

Wesie, 2017-07-08:
In the meantime, I've had the both TX once again in Frankreich of me. 
I update  both TX to version 2.2.0 
The Problem is still available. 
The 3rd TX- also an old device : a 9XR with an Atmega128 was flashed also from the Turnigy SW to Opentx 2.2.0 
After start the TX and Eeprom initialization (eeprom Warnung after first start) I read the eeprom in Companion : Same message, but companion convert the file. I save the file, close all files in companion, reopen the modellfile and flashed back into the TX. 
If I start the TX, the eeprom warning occurs again, the eeprom must initialized once again and all Models are lost. This step is reproducable to 100%. Only chance to use the TX is to edit the models in the TX by himself by hand 
In the meantime all 3 TX are back to my friend in the other corner of Germany

closed 2019-08-19 by kilrah