opentx/opentx #5035: Taranis - "Sensor Lost"

carloscampos775, 2017-06-17:
Using OpenTX 2.2 the Taranis will sound intermittent warnings about a "Sensor Lost". After going through the sensors, I found the issue is with the VFAS sensor. The VFAS sensor will stop updating at random intervals. I have also discovered this issue is only happening on FCs that use a current sensing pin like the "SP Racing EVO - Betaflight" and "Revolt F4 - Raceflight". This issue is not happening on boards like the Betaflight F3 or Kiss FC that take direct battery voltage.

3djc, 2017-06-17:
This is an issue on their side. OpentTx expect a regular stream of sensor data. If one fails to send for several seconds, a warning is raised.

closed 2017-06-17 by 3djc

carloscampos775, 2017-06-17:
This was not happening prior to the OpenTX 2.2 update.

schwabe, 2017-06-17:
It probably was happening before. The warning that a sensor stopped sending is new in 2.2.

carloscampos775, 2017-06-17:
I understand that this might have been happening before. This new feature
has cause an issue where there was none prior. There should be an option to
disable the “sensor warning alert” for all sensors and/or just a particular
sensor.

3djc, 2017-06-17:
There was one before, you where not getting data correctly, you just didn't know about it

schwabe, 2017-06-17:
@carloscampos775 the timeout is 15s. If a sensor does not give you a reading every 15s something is very broken.

kilrah, 2017-06-17:
No, it simply revealed a likely bug with the software you're running on your flight controller. So you should contact the maintainer of that software so they can fix it.

carloscampos775, 2017-06-17:
This is only happening 1 to 3 per flight. The sensor is always regained
directly after the warning message. I will post the issue with the FC.

rondofpv, 2017-08-04:
im having the same issue!

BarryCole, 2017-10-12:
I get this issue too, and I am only using Frsky Vario, and GPS modules.