opentx/opentx #5036: opentx 2.2 logical switches chrono

FlipAirboost, 2017-06-17:
Logical switch panel,
 chrono function,
 the time programmed in the duration column is no longer taken into account with opentx 2.2.0
Same parameters worked fine with opentx 2.0.20.
[image]

FlipAirboost, 2017-06-18:
Version 2.2 speaks more slowly and is only able to announce time every 2 seconds with the same sound pack "amber as version 2.0.
Open tx 2.0 can announce time almost every second.

projectkk2glider, 2017-06-24:
Your first problem is not understandable. Your picture does not correspond with the description provided for L2. 

Your second problem: you can use another sound pack that has shorter (ie faster) sounds. This is not a bug.

FlipAirboost, 2017-06-24:
Hi projectkk2glider,
First problem: I have updated my first post and here are two videos.
I edit the timer to shorten the videos but you can see Lxx highlight it.
https://drive.google.com/file/d/0BzFhb9hmRA8Pb2NSTEFSNzRudEk/view?usp=sharing
https://drive.google.com/file/d/0BzFhb9hmRA8PNnFqbHJuM0gxbDA/view?usp=sharing

Secondly, same sound files need same start up playing time to be read simultaneously , in my opinion.
Why opentx 2.2.0 can't handle it?

kilrah, 2017-06-24:
> Version 2.2 speaks (I will say read)more slowly and is only able to announce time every 2 seconds with the same sound pack "amber as version 2.0.

There should not be any change there.
You DO need different sound packs for both so the difference is most likely there.

FlipAirboost, 2017-06-24:
Wat a pity, I love Amber voice!
[image]
Amber Sound files are lighter yet.So there is more than a difference in name length and filename?
Here are my renamed system sound files.
https://drive.google.com/drive/folders/0BzFhb9hmRA8POXZJVFlSTFJDdkk?usp=sharing
Regards Philippe

FlipAirboost, 2017-06-25:
Kilrah says
> There should not be any change there, it isn't even possible for a speed difference to happen.
>  You DO need different sound packs for both so the difference is most likely there.
O.K. I tried it with the official sound packsdcard-taranis-x9-2.2V0010 , same problem, there is a restriction in the pipeline,similarly, on the simulator and TX
video here:
https://drive.google.com/file/d/0BzFhb9hmRA8PVmNUYzVaam1yMm8/view?usp=sharing
Regards Philippe

kilrah, 2017-06-25:
> https://drive.google.com/file/d/0BzFhb9hmRA8Pb2NSTEFSNzRudEk/view?usp=sharing
https://drive.google.com/file/d/0BzFhb9hmRA8PNnFqbHJuM0gxbDA/view?usp=sharing

There seems to be a difference in behavior between 2.1 and 2.2 indeed, but 2.2 is more correct.
If you check well, the announcement on 2.1 does announce every second but it gets late becasue the files take too long to play, so much that at 8 seconds it's actually one second late and skips sstraight from 8 to 10. 2.2 doesn't do that and if it's too tight it doesn't play.
My personal sound pack has shorter files and they do play each second on 2.2 (until it gets to longer files like 23).

All normal then.

FlipAirboost, 2017-06-25:
>but 2.2 is more correct. 
I agree with that.
> All normal then
No, the logic switches do not work as expected, as you can clearly see on video.

kilrah, 2017-06-25:
I only referred to the speech.

Still not understanding your point about the LSs. You say 

>the time programmed in the duration column is no longer taken into account at the right time with opentx 2.2.0

That doesn't really mean anything.... what do you mean by "at the right time"??

kilrah, 2017-06-25:
As expected, XY Problem.

Your actual issue is "Timer 1 doesn't stay at 0 during the time the logical switch that triggers the Reset Timer 1 is active".

This has nothing to do with logical switches, they work perfectly fine. The Reset special function was however changed from being active as long as the switch is on to only be triggered once when the switch is turned on. 
This was an undesired side effect of a fix for another problem and will be reverted in 2.2.1. 

In the meantime you can use "Set Timer1 00:00:00" instead.

closed 2017-06-25 by kilrah

projectkk2glider, 2017-06-25:
I will reference issue #3858 here, if we ever need to find it, because its solution caused this issue.

FlipAirboost, 2017-06-25:
Thank you all,
It can remain as it is once you know that.
I updated the first post.
Regards 
Philippe

commit a587f8b8 (2017-06-25): Edge detection only on "Reset Flight", commit 6bc6e074e17e28138fcb23b1e6a3e91eef0b406a introduced unnecessary change in behavior for Reset Timer (fix #5036)