opentx/opentx #5037: TARANIS X9E - menu wheel (encoder) trouble

bandurat, 2017-06-17:
Hi, I have a trouble with menu wheel (encoder).
On bootloader mode everything is OK, the menu wheel works good, but on "normal mode" sometimes the menu wheel is frozen, or it works strange.
I tried a firmware change and factory reset, but it did not solve the problem.
What should I do with this?
Thank You very much.

Here are the videos:
https://www.youtube.com/watch?v=9BbKqx2vFww
https://www.youtube.com/watch?v=h5Sbo8DU6Gg

3djc, 2017-06-17:
I'm using an x9e daily, never experienced anything like that. Any lua scripts running?

bandurat, 2017-06-17:
No, I have no lua script.
It is a new radio I bought it one mounth ago. But I think it is not a hardware problem, because on bootloader mode everithing is OK. :/

kilrah, 2017-06-17:
Maybe mention your firmware versions....?

bandurat, 2017-06-17:
Of course, now I have 2.2.0. But with 2.1.6 or 2.1.9 I had a same problem. On videos You see 2.2.0 FW with 2.1.6 bootloader, but I also upgraded bootloader to 2.2.0 with no result.

kilrah, 2017-06-17:
Can't think of anything but flaky hardware since no one else has that problem.

bandurat, 2017-06-18:
Likely yes, but I don't understand why it is OK all the time in bootloader mode... :/

kilrah, 2017-06-18:
The bootloader might read the keys at a different speed or in different order, and that doesn't reveal the issue...

schwabe, 2017-06-18:
Might be something simple like a loose ground connection. That would also explain that behaviour but I agree, there is something fishy with your hardware.

projectkk2glider, 2017-06-24:
Closing since this appears to be a hw issue. I have two radios with encoder, none of them has this issue.

closed 2017-06-24 by projectkk2glider