opentx/opentx #5038: Issue with Taranis X9D+ A01 and the "bindopt" compilation option

LapinFou, 2017-06-18:
Hi folks,

I order a XSR RX in order to use it as a buddy box wireless system using its CPPM output.
I tried to bind the XSR and disable its telemetry using the **bindopt** compilation option.
To achieve this (after several tries), I discovered that the following requirements must be respected to make it work:
1. Flash the Taranis with the **bindopt** compilation option.
1. Flash the XSR with the latest firmware.
1. Flash the XJT internal module with the latest firmware (if not, the RX will bind, but **the telemetry will not be disabled**).

Now, here is my issue:
my X9D+ is a **A01 version** meaning the RF board doesn't have the SWR/RAS hardware soldered on. Flashing my Taranis with the latest XJT firmware will generate a **defective antenna** error message. The RAS is reading 98/99. 
Rolling back the internal XJT firmware to the version ``XJT_141016.frk`` fixed the **defective antenna** error message (RAS continuously read 0), but then the **bindopt** feature doesn't work.

Not sure if it is a OTX or FrSky issue, but any help will much appreciate. 😃

3djc, 2017-06-18:
Frsky issue I'm afraid,  nothing we can do.

LapinFou, 2017-06-18:
😭 
No magic option on OTX side to coerce the RAS/SWR value to 0?

schwabe, 2017-06-18:
We try to detect to the defective version: 
```
  #define IS_SWR_VALUE_VALID()         (telemetryData.xjtVersion != 0 && telemetryData.xjtVersion != 0xff)
```

kilrah, 2017-06-18:
Which "latest" has the problem? Bind options are in since the XJT firmware that came out in December or so.

schwabe, 2017-06-18:
we check if the version reported is known bad (  #define IS_SWR_VALUE_VALID()         (telemetryData.xjtVersion != 0 && telemetryData.xjtVersion != 0xff)

kilrah, 2017-06-18:
I'm asking @LapinFou what version he tried that didn't work. AFAIK they recently released a new XJT version that would be the "latest", but he could use the one before that and have bind options already.

schwabe, 2017-06-18:
I build a version that displays xjt version (where RAS is displayed): 

https://github.com/opentx/opentx/tree/schwabe/display_xjtver

Compiled binary http://plai.de/opentx/x9d+-display-xjt.bin

Can you check what versions are displayed on your module? Mine (X9D+) is displaying 0x200 on the december firmware.

LapinFou, 2017-06-18:
Hi folks,
Thanks a lot for helping!!!
The current version I'm running is ``XJT_141016``. But, as previously mentioned, I can't disabled the telemetry.
Then, I installed the version ``XJT_FCC_170317`` which does work for disabling the telemetry, but gives me the **defective antenna** error.

@schwabe I will try you version.

3djc, 2017-06-18:
Well until frsky fix there thing, can't you bind with the defective version and switch back to a working one after that? (the options should remain)

LapinFou, 2017-06-18:
@schwabe 
I must look in the main menu page 7/9 bottom line, correct ?

@3djc 
I didn't think about this one.

kilrah, 2017-06-18:
XJT_161214 was the first to get the binding options. Never had a report of an issue with that.

LapinFou, 2017-06-18:
@schwabe 
``XJT_130725`` gives RAS 97 XJTVER 1025 + **Antenna Defective** error
``XJT_141016`` gives RAS 0 XJTVER 0
``XJT_161214`` gives RAS 96 XJTVER 1025 + **Antenna Defective** error
``XJT_170317`` gives RAS 98 XJTVER 1025 + **Antenna Defective** error

@kilrah 
Please note that my X9D+ is an A01 version (not SWR hardware soldered on).

LapinFou, 2017-06-18:
For testing, I tried with OTX2.1.9
Same result as with OTX2.2.0

schwabe, 2017-06-18:
@LapinFou the test is identical in 2.1.9. The question is if the version 1025 is specific to the A01 hw revision or not.

LapinFou, 2017-06-18:
@schwabe 
I tried on the Taranis of my son mainly used for wireless buddy box. I personally also own a X9D+ A01. If your are 200% sure than changing the XJT firmware then going back to the  ``14106`` doesn't affect the bind, then I can try. I want to avoid to be force to re-bind all my models for this test.

kilrah, 2017-06-18:
XJT firmware has never affected bind as far as we know.

LapinFou, 2017-06-19:
@kilrah 
Ok then, I will try the modified firmware on my A01 x9D+.

LapinFou, 2017-06-22:
@schwabe 
Here is the result with a 2nd A01 Taranis X9D+
``XJT_130725`` gives RAS 48 XJTVER 1025 + No **Antenna Defective** error
``XJT_141016`` gives RAS 0 XJTVER 0
``XJT_161214`` gives RAS 48 XJTVER 1025 + No **Antenna Defective** error
``XJT_170317`` gives RAS 48 XJTVER 1025 + No **Antenna Defective** error

LapinFou, 2017-06-24:
I send an email to FrSky the 18th of June. Still no answer. Here is the email I send, maybe you do have a more direct channel than me to talk to FrSky??
```Hi folks,

I got 2 Taranis X9D+ version A01 (production date 05/09/2014).
For some reason I want to use a XSR RX without telemetry. So I followed the following procedure:
1. Installed OTX 2.2.0 with the "bindopt" option.
2. Flashed the XSR with the latest FCC firmware version.

The bind was successful, but the telemetry was still enabled.

After some tries, I discovered than the internal XJT of my X9D+ must be flash with the latest firmware (version 170317).

Doing so, make everything work as expected. I successfully bind my XSR without telemetry.

However, my X9D+ is an A01 version. So, there is not SWR/RAS soldered on the RF board (not a problem, since RSSI is the good data to looking at).
But, the latest XJT firmware generated a "defective antenna" error. Indeed the SWR/RAS is not reading anymore 0, but 98 or 99.

I opened a ticket on the OpenTX gitbug, bu they told me than it is a FrSky problem.
https://github.com/opentx/opentx/issues/5038

Is it possible to fix the problem??

BR,
Seb
```

closed 2017-11-08 by kilrah