opentx/opentx #5040: Can't download FW

diggit, 2017-06-18:
Companion22 reports "Invalid firmware" when it attempts to download FW for my X7.
I can read download link from stdout [`http://downloads-22.open-tx.org/nightly/firmware/getfw.php?fw=opentx-x7-eu-noheli-lua-bindopt-massstorage-sqt5font-en.bin`](http://downloads-22.open-tx.org/nightly/firmware/getfw.php?fw=opentx-x7-eu-noheli-lua-bindopt-massstorage-sqt5font-en.bin)

When I try to visit this link in browser, I get `ERROR:5` response.

Is FW building service down?

kilrah, 2017-06-18:
You'er probably still using a companion RC, you must update to 2.2.0 release.

closed 2017-06-18 by kilrah

diggit, 2017-06-18:
Thanks for fast reply!
Updated companion to version 2.2.0 and everything works well.