opentx/opentx #5042: Throttle warning on different switch

kennlicht, 2017-06-19:
Hello to all. I'm new to this forum and here is my first question:

I don't use the throttle stick but a switch (SA) instead to start the motor of my model. I know I can configure a "switch warning" for that switch. But I would really appreciate if I could also get a "throttle warning" if the switch is not in its idle position.

I haven't found a way to configure that behaviour in OpenTX 2.2... Is it possible though?

3djc, 2017-06-19:
Sorry, this is not a forum, but a bug tracking system. Your question should be asked on forums like http://openrcforums.com/forum/viewforum.php?f=45 or rcgroups : https://www.rcgroups.com/forums/showthread.php?2823315-OpenTx-2-2

closed 2017-06-19 by 3djc

kennlicht, 2017-06-19:
Ok. Supposed I found a bug... Thank you for that incredible fast response!