opentx/opentx #5048: Cant Select Sensor in the calculated sensor menu

merinkumar, 2017-06-20:
Hi, Am using opentx 2.2 on both the simulator and qx7 radio, am not able to add an calculated current sensor, please find the attached screen shot. The sensor just displays --- but cant select " curr. " Please help

[image]

kilrah, 2017-06-20:
This only shows actual current sensors, that have been detected properly so the unit is set right.

This is not a place for usage questions.

closed 2017-06-20 by kilrah

olyrhc, 2017-06-21:
I've also had this issue with OpenTX 2.2.

I can create a new calculated sensor in Companion and set the source-sensor to "Cells" (for example) and it works fine. If I try to duplicate it in OpenTX 2.2, it refuses to let me select the "Cells" sensor, even though it's been detected properly.