opentx/opentx #5049: Setting Gvar with encoder X12S, X9E, QX7 (or with +/- Switch at X9D)

hrenz, 2017-06-21:
Hy, 
only an Idea, working with Gvars and Poti settings.

I can set Gvar value with Potis, ok, but next time  poti is at an other position, not so fine.
I can set Gvar value with a  3 way toggle switch, and gvar value next time is the same, thats fine,ok
(But there is not a real 3 way toggle switch at all Frsky Tx)

I think it should be possible to set the Gvar with the encoder wheel  at X12S, X9E and QX7 
(or with the + and - switches at X9D)

MPX hat this funktion with the encoder wheel too

kilrah, 2017-06-21:
This is difficult because the encoder always has a "direct" action on the UI. 
Would need a whole mechanism to lock the UI whenever that function is active...

3djc, 2017-06-21:
You can auto record pot positioning so that it will ask you to put it back in the same position as when you last used this model

hrenz, 2017-06-23:
ok, next Idea for an encoder adding
X9E, X9D, Is  it possible to use a encoder at a 3 Pos Switch socket
at openTx Hardware Setup we need a encoder setting instead of 3 Pos Switch setting

Link like this: https://www.mikrocontroller.net/articles/Drehgeber

closed 2019-11-15 by stale[bot]