opentx/opentx #5050: Companion app on OS X missing avrdude.conf

bdemers, 2017-06-21:
Version: v2.2.0
OS: macOS 10.12.5

I installed the OpenTx Companion app and attempted to backup my radio's current config.

The error message showed that avrdude could NOT found the find '/usr/local/etc/avrdude.conf' (as this did not exist on my computer.

Hack work around: `brew install avrdude`

I'm assuming that the Companion should be bundling this file and setting the `-C` option when running avrdude?

thirsch, 2017-07-29:
I can confirm the issue. After referencing a valid avrdude.conf, it turned out that the build seems to have no libusb support:

```bash
avrdude: Version 6.3, compiled on Dec  6 2016 at 22:51:09
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "./avrdude.conf"
         User configuration file is "/Users/horst/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: error: no usb support. Please compile again with libusb installed.

avrdude done.  Thank you.

```

schwabe, 2017-08-01:
Strange:
Definitively linked to libusb, are you sure the version from the dmg is called?

```
[16:26]arne@pan:~% otool -L =avrdude 
/usr/local/bin/avrdude:
	/usr/local/lib/libusb-1.0.0.dylib (compatibility version 2.0.0, current version 2.0.0)
	/usr/lib/libobjc.A.dylib (compatibility version 1.0.0, current version 228.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1226.10.1)
	/usr/lib/libedit.3.dylib (compatibility version 2.0.0, current version 3.0.0)
	/usr/lib/libncurses.5.4.dylib (compatibility version 5.4.0, current version 5.4.0)

```

closed 2017-08-04 by kilrah

bdemers, 2017-08-04:
Sweet! :+1:

thirsch, 2017-08-05:
@schwabe strange indeed. Here is the output from otool and again the call to avrdude. Both from the dmg package:

```
Berlin:Resources thomas$ pwd
/Applications/OpenTX Companion 2.2.app/Contents/Resources
```

```
Berlin:Resources thomas$ otool -L ./avrdude
./avrdude:
	@executable_path/../MacOS/libusb-1.0.0.dylib (compatibility version 2.0.0, current version 2.0.0)
	/usr/lib/libobjc.A.dylib (compatibility version 1.0.0, current version 228.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1226.10.1)
	/usr/lib/libedit.3.dylib (compatibility version 2.0.0, current version 3.0.0)
	/usr/lib/libncurses.5.4.dylib (compatibility version 5.4.0, current version 5.4.0)
```

```
Berlin:Resources thomas$ ./avrdude -c usbtiny -p m64 -C ./avrdude.conf -P usb -U flash:r:/tmp/20170805.hex:i
avrdude: error: no usb support. Please compile again with libusb installed.

avrdude done.  Thank you.
```

darrenallen85, 2020-05-16:
Work around for those that strike this 
1. brew install avrdude
2. cd /Applications/OpenTX Companion 2.2.app/Contents/Resources
3. mv avrdude avrdude.org
4. ln -s /usr/local/bin/avrdude avrdude

good luck

commit 7ee4bf90 (2017-08-01): Bundle and use avrdude.conf on OS X (closes #5050)
commit bd968bd6 (2017-08-04): Bundle and use avrdude.conf on OS X (closes #5050) (#5127)