opentx/opentx #5053: Throttle warning on different switch

kennlicht, 2017-06-23:
Because I'm now convinced that this is a bug or at least a source of improvement, I would like to reopen issue #5042.

Here is a screenshot of what I'm doing:
[image]

I've configured the "Throttle Source" to CH04, which is mixed from switch SA and not from the throttle stick. Timer 2 reacts on that source as expected. All ok up to this point.

Additionally I turned the option "Throttle Warning" on. Now I get this warning every time when turning my radio on with the throttle stick not in its lowest position. But this warning is not related to my switch SA respectively to CH04. I think it should!

OpenTX still warns me with a switch warning, but it's likely to overlook the throttle position (SA in my case) and ignore the warning by pressing a button. A second warning like the throttle warning would be really nice.

powercycle, 2017-06-24:
What if a dialog box was displayed based on a special function?  

Could this currently be accomplished with a lua script maybe?

jtaylor2, 2017-06-24:
>What if a dialog box was displayed based on a special function?
>Could this currently be accomplished with a lua script maybe?

A throttle warning needs to be displayed before the rf is turned on or the throttle channel enabled.  I don't know LUA, but doubt it is available that early in the startup.

3djc, 2017-06-24:
It's not, and LUA cannot 'jump' at your screen either, it needs to be called manually

kennlicht, 2017-12-15:
Seems to be fixed in 2.2.1 RC3. Perfect!

closed 2017-12-15 by kennlicht