opentx/opentx #5054: Companion 2.2.0 won't start on Win 10

gofaster, 2017-06-23:
Just downloaded and installed on my system running Win 10 Home 64bit.  Version 10.0.15063 Build 15063
Companion 2.2.0 will not launch, just silently fails.

Window Event Log entries:
Faulting application name: companion.exe, version: 0.0.0.0, time stamp: 0x592dc1d1
Faulting module name: Qt5Core.dll, version: 5.7.1.0, time stamp: 0x583ffb6c
Exception code: 0xc0000005
Fault offset: 0x00109e46
Faulting process id: 0x39e0
Faulting application start time: 0x01d2ec4a40733842
Faulting application path: C:\Program Files (x86)\OpenTX Companion 2.2\companion.exe
Faulting module path: C:\Program Files (x86)\OpenTX Companion 2.2\Qt5Core.dll
Report Id: 9c6278d3-daab-47c0-bbcb-52951381dd4b
Faulting package full name: 
Faulting package-relative application ID: 

...and
Fault bucket 108952813997, type 1
Event Name: APPCRASH
Response: Not available
Cab Id: 0

Problem signature:
P1: companion.exe
P2: 0.0.0.0
P3: 592dc1d1
P4: Qt5Core.dll
P5: 5.7.1.0
P6: 583ffb6c
P7: c0000005
P8: 00109e46
P9: 
P10: 

Attached files:
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER7CBE.tmp.dmp
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER7D2C.tmp.WERInternalMetadata.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER7D3B.tmp.csv
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER7D5B.tmp.txt

These files may be available here:
C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_companion.exe_3cdbda0c957e2d52add7cab4083f48642be4_4bb22eff_131081bf

Analysis symbol: 
Rechecking for solution: 0
Report Id: 9c6278d3-daab-47c0-bbcb-52951381dd4b
Report Status: 268435456
Hashed bucket: eb97cadf597855d73b1336f669d7dc23

gofaster, 2017-06-24:
Well, I removed and reinstalled again - no go, silently fails to start.  Installed 2.1.19 as a test, works fine.  Reinstalled 2.2 again with removing first, it runs!   Wierd.

closed 2017-06-24 by gofaster

Airheadbit, 2017-07-13:
Any fix for this? I tried to configure 2.2 for Windows 8, 7, XP and all fail

Airheadbit, 2017-07-13:
Why is this issue closed? was there a fix?

gofaster, 2017-07-13:
Per my prev comment, I installed 2.1.19 to test if it fails as well.  That started up fine. Then I ran the 2.2 installer again for a 3rd time.  Companion started up with no problems. No idea why and unable to replicate so I closed the issue.