opentx/opentx #5055: Lua script causes q x7 to reboot

Jmelz, 2017-06-23:
I was attempting to use this script for telemetry purposes, and found that it crashes and reboots radio upon execution.

I believe it is related to the draw commands, but I am not certain. Removing several commands and restructuring the script made it work on the x7, but I wanted to report it.

[dlvector-v1_1.lua.txt](https://github.com/opentx/opentx/files/1098838/dlvector-v1_1.lua.txt)

https://www.rcgroups.com/forums/showpost.php?p=34749134&postcount=49435

projectkk2glider, 2017-06-24:
OK, found the problem. But you might wanna fix the script it calls `lcd.drawFilledRectangle` with a height of 200, which is obviously too big (I think LCD has 64 lines) (x=0 '\000', y=0 '\000', w=33 '!', h=200 '\310', pat=255 '\377', att=255)

projectkk2glider, 2017-06-24:
@bsongis tough question for you :stuck_out_tongue: Do you remeber why `y` coordinate type was changed from unsigned to signed in https://github.com/opentx/opentx/commit/2b4823ab844818a47c0447a9215ede204aafd456#diff-4cfa7f631a434f02243db59e2dc7b2c0R363 ?

bsongis, 2017-06-24:
I think because otherwise negative numbers do overflows

Jmelz, 2017-06-24:
I have already edited the script to work, but someone at RC Groups suggested I report as a bug. Thanks for your great work on opentx!

Separately, I am trying to figure out how to get the GPS telemetry sensor value to stay on screen even once telemetry is lost.

3djc, 2017-06-24:
You simply do not update the results if the function does not return a table

Jmelz, 2017-06-24:
Thanks - that was so easy, even I could do it! :) 

 --  gpsValue = "not currently available" -- commented out as not to update with useless info :)

projectkk2glider, 2017-06-24:
You were right to report the issue, this kind of issues need to be fixed.

projectkk2glider, 2017-06-25:
Interesting, when you take this expression (y is `scoord_t == int8_t` and h is `coord_t == uint8_t`):
```
    for (scoord_t i=y; i<y+h; i++)

```

Which type is used in this comparison ` i<y+h`  ? :open_mouth:  

Well `y+h` is converted to `int` after the addition, here are couple of examples for different y and h values:
```
(gdb) p (unsigned char)5 +(int8_t)(7)  
$3 = 12
(gdb) p (unsigned char)5 +(int8_t)(-7)  
$4 = -2
(gdb) p (unsigned char)200 +(int8_t)(127)  
$11 = 327

```
All good until the last example!. 

Now comes our example from this bug: `y = 0`, `h=200` which are perfectly good values and fall withing their respective type range:
```
(gdb) p (unsigned char)200 +(int8_t)(0)  
$15 = 200
```
So the condition becomes `i<200` which can never be **false**!, remember max i = 127.

One possible solution is to cast the result of the addition:
```
for (scoord_t i=y; i<(scoord_t)(y+h); i++)
```

PR #5058 (2017-08-15): Fixes #5055: infinite loop in lcdDrawFilledRect() when `h` parameter …
…is bigger than 127. Similar fixes in other places.

Closes #5055
commit a167e9f7 (2017-08-15): Fixes #5055: infinite loop in lcdDrawFilledRect() when `h` parameter is bigger than 127. Similar fixes in other places. (#5058)
closed 2017-08-15 by bsongis