opentx/opentx #5057: Companion suggestion to supresss unused FMs from inputs and mixes lists

elecpower, 2017-06-25:
Just a cosmetic suggestion. Currently unused FMs i.e. those without a switch, display in the lists when one or more FMs are unchecked for a input or mix line. FM0 would always display.

mpjf01, 2017-07-03:
This seems to be a change to what every user of flight modes will see and a change to the paradigm. Right now if you tick a FM box in a mix it appears in a mix line whether used or not. Users expect that. It's useful when de-bugging complex setups. It gives a definitive statement of which mix(es) a flight mode is active in and which it is not. This is important. It's more than a cosmetic suggestion.

elecpower, 2017-07-03:
@mpjf01 I take your point however if a flight mode does not show on a mix or input line it highlights that either the user does not want it to apply to one or more flight modes or the flight mode has no true condition and thus the line would be ineffective.
A mix line showing that it applies to a flight mode but the flight mode will never be active is also misleading.

mpjf01, 2017-07-03:
I don't agree. Leaving something blank is not going to "highlight" anything. Showing the FM in the mix is showing that the user has selected it in the mix, a deliberate decision. It isn't showing that the FM has been made active. However if you adopt the strategy of only naming FMs that you use it's instantly clear which FMs are in use in any mix and which are not anyway.

I'm not sure what problem is trying to be solved here. We are all used to the way it is shown now, changing it will create confusion and unnecessary user interactions - most of which if the past is any guide will be negative. If there is a problem I do not believe the benefits of change outweigh the costs. But as I don't make the decisions here it hardly matters what I think.

elecpower, 2017-07-03:
Our discussion just shows that each user has the potential to use the product in different ways. That does not mean either side is wrong.
An extension of the idea could be the addition of a user maintainable parameter to suppress or not.

lshems, 2017-07-03:
This change would stop you from changing mixes without having the FM activated. So you need more unnecessary user interaction. 

Also, since there will always be a default FM, undesired behaviour is bound to happen. Which FM is the default?

elecpower, 2017-07-03:
Not proposing any change the mix or input edit forms. All FMs would be available. No changes to FMs ie having to name. Just suppress from lines in mixes or inputs lists if a FM is selected in mix or input but no condition defined to activate the FM except for FM0 where no switch is ever defined. FM0 would be suppressed when unchecked.

[image]

[image]

[image]

kilrah, 2017-07-04:
It is a desire of the project not to hide anything whether it's relevant or not. People are used to this and rely on it, so something like this isn't very desirable.

It's common for people to temporarily disable things by removing the switch, but they'd still want to see what's going to happen when they'll add it back.

elecpower, 2017-07-04:
IMHO this principle is not being applied consistently for example Special Functions where field values are hidden when the switch is changed to none and reappear as they were when a switch is assigned. To a lesser extent Build Options affect what is displayed in Companion.
If the view of the dev team is as you state then close the issue and reject the PR.

lshems, 2017-07-07:
Good point. Special functions shouldn't hide if no switches assigned. It is confusing.

Even model conversions and eeprom saves do not work properly in that perspective.

When using a FM as switch on a voice calling the FM, and you want to disable this temporarily because of annoyance during repetitive launch practice for instance, it should remain visible, and certainly not being deleted on eeprom save or things like that.

elecpower, 2017-07-13:
No support for suggestion so closing

closed 2017-07-13 by elecpower