opentx/opentx #5064: Close Window Keyboard Shortcut Wrong on Mac

xtrmtrk, 2017-06-27:
On the Mac version, Command-W should close an open window when editing a model, not Escape.  For those of us used to the Mac conventions, this switch is frustrating.

projectkk2glider, 2017-07-24:
Just a reminder: when fixing this first check if perhaps the issue is caused by the Qt library.

projectkk2glider, 2017-07-24:
@xtrmtrk did you had the same problem in previous versions - ie 2.1?

xtrmtrk, 2017-07-24:
Yes, there's been the same behavior for as long as I can remember (I'm patient, or lazy - the frustration took years to build for me before I submitted a bug!).

What's kind of odd is that Command-W works correctly for a file window - the window that lists all of the models.  But it does not work when trying to close individual model windows or simulator windows.

raphaelcoeffic, 2018-05-07:
@xtrmtrk that's probably because the "open windows" are considered dialog boxes, which are not closed by Cmd-W either on Mac.

closed 2019-08-14 by raphaelcoeffic