opentx/opentx #5065: Companion 2.2.0 GVARs not simulated

jechas01, 2017-06-27:
OS: Arch Linux
Version: 2.2.0-2

Issue: GVARs never seem to update in the simulator. They're always 0.

Steps to reproduce: Create a new model, then add a special function as such:
[image]

Then go to the simulator, click View -> Radio Outputs (to get the gvar display), and then slide RS around.

[image]
[image]

Expectation: The gvar moves with RS
What I'm seeing: It's just 0.

I noticed this when I updated from 2.1 and my models suddenly stopped behaving as expected. When written to my x9d+, everything looks fine. It's just the simulator that's being weird.

3djc, 2017-06-28:
What was the installer you used ? Because it does work fine.

jechas01, 2017-06-28:
It was built using this AUR package: [opentx-companion.tar.gz](https://github.com/opentx/opentx/files/1109899/opentx-companion.tar.gz)

Actually, looking at [the comments on the package](https://aur.archlinux.org/packages/opentx-companion/), the maintainer mentions getting the simulator to work "soon." Is there something incorrect about the build method in the PKGBUILD?

3djc, 2017-06-28:
It is not built by us, and this unofficial version is not built correctly . We already raise awareness about that, and recommended for the maintainer to look at our build scripts

closed 2017-06-28 by 3djc

bsongis, 2017-06-28:
@3djc wait, I am on Manjaro, I can have a look, I didn't know there was already something done in the AUR!

jechas01, 2017-06-28:
Just looking through the wiki, I might have found the issue. Is the missing `-DGVARS=YES` flag the culprit?

I'm almost done recompiling now - will report back with the results when it finishes.

3djc, 2017-06-28:
Here is how we build : https://github.com/opentx/opentx/blob/2.2/tools/build-companion-release.sh

jechas01, 2017-06-28:
@3djc Thanks, I'll see how hard it is to adapt that to an arch-friendly package