opentx/opentx #5066: Companion 2.2: MacBook Trackpad Scrolling Very Difficult in Simulator

xtrmtrk, 2017-06-28:
When using the Companion Horus simulator on a MacBook, using the trackpad to activate the Horus scroll wheel (inside MDL/TELE/RTN/SYS buttons) is extremely difficult.  Using the Mac's trackpad scrolling gestures (2-fingers) results in an almost uncontrollable stopping and jumping of the selected area in the simulator.  It's practically impossible for example to configure items when adding LUA widgets.

Even if this behavior was fixed, it would also be nice if there was an alternative way to control the small scroll wheel in the simulator (e.g. command+left/right arrow keys).

schwabe, 2017-06-28:
press the help button there are alternative keys defined.

closed 2017-06-28 by schwabe

xtrmtrk, 2017-07-01:
Perfect, thanks!