opentx/opentx #5067: "convenient mode" missing from S6R Lua script

Jmelz, 2017-06-29:
On June 5th, FRSKY introduced a new firmware with a "simple" or "convenient" mode that only allows for 2 flight modes and "off".

https://www.rcgroups.com/forums/showpost.php?p=37669372&postcount=3106

The Lua script appears to be missing the checkbox to enable this functionality. The PC app has a checkbox called "convenient mode" while FRSKY referred to the feature as "simple mode".

In other words, if a user updates the firmware on an s6r with this firmware file, they are still unable to select convenient mode by using the lua script.

3djc, 2017-06-29:
There is no such official firmware : http://www.frsky-rc.com/download/view.php?sort=&down=393&file=Firmware-%20S6R

Jmelz, 2017-06-29:
hi 3djc,

correct, is not official, but it was released by FrSky and communicated from their official account on rcgroups.

I'm not sure what they wish to do with this file, but I am just pointing out that the functionality does not yet exist in the lua script.

kilrah, 2017-06-29:
And it can't exist if we're told nothing about how to make it work and it's not even available to reverse-engineer...

Jmelz, 2017-06-29:
Completely understand - I've contacted FrSky and asked them to commit the fw as official so that the lua script can be updated.

Jmelz, 2017-07-01:
FrSky said they will be reaching out soon. They have now placed the firmware on the website at least:

http://www.frsky-rc.com/download/view.php?sort=&down=394&file=Beta Firmware-S6R

closed 2017-07-20 by kilrah

3djc, 2017-08-04:
[image]

[image]

Jmelz, 2017-08-04:
Very interesting... I reflashed my receiver with the zip file I pasted. What version are you using?

3djc, 2017-08-04:
S6R_FCC_170615.frk

Jmelz, 2017-08-04:
Very strange indeed. I confirmed I'm using the SxR.lua for the x7... I reflashed fw again, and no dice. I'll test some more. Thanks for your quick reply and photos 👍

PR #5099 (2017-07-20): Review SxR script to handle lastest receiver firmware revision
- Add 'quick' mode compatibility
- Add compatibility with previous receiver firmware (not blocking anymore, unsupported field/functions just keep indicating  '---')
- rename S6R to SxR to acknowledge S8R is supported too

This also fixes #5067