opentx/opentx #5069: QX7 drawPixmap enable ?

aurelx03, 2017-07-01:
Hi,
I would like know if its possible to enable drawPixmap for the QX7, i know resolution of qx7 screen is small but need to upload small image for custom screen script or other. It is a pity to disable a function between 2 radios.

Kiss friends <3

3djc, 2017-07-01:
Not only is the resolution lower, but it does not do greyscale either, it is pure 1 bit bitmaps, not helpful at all

aurelx03, 2017-07-01:
yeah but for example you would draw an icon or a splash screen for your script u must draw the picture pixel by pixel its annoying

rdeanchurch, 2017-07-13:
One bit on the low res screen is not perfect, 
but it beats nothing by a country mile.
Please consider putting drawbitmap for Qx7.
[image]

RCdiy, 2017-07-30:
Better than nothing. At least it gives us a choice to live with the image or not.

3djc, 2017-08-06:
Done in https://github.com/opentx/opentx/commit/541aa459cf5e368dbff916553e4124abc43e2fb9

closed 2017-08-06 by 3djc