opentx/opentx #5070: Simulator not playing Tmp1 and Tmp2 values

Riffer, 2017-07-01:
The values in the attached log (renamed vom csv to txt to upload here) are from inav - Tmp1 and Tmp2 are not replayed in the simulator in Version 2.2.0 or N371 or any other version I could get my hands on.

[Arkai_Wing-2017-01-24.txt](https://github.com/opentx/opentx/files/1117353/Arkai_Wing-2017-01-24.txt)

mpaperno, 2017-07-01:
The telemetry log player has some issues.  One of them is that the log column headings are expected to be in a [particular format](https://github.com/opentx/opentx/blob/2.2/companion/src/simulation/telemetrysimu.cpp#L626) (and always in English). For now you should be able to work around the Tmp1/2 issue by manually editing the column headings to read `Tmp1(@C)` and `Tmp2(@C)` (so, adding the `(@C)` to each header).  

Naturally, do this in a plain-text editor. If you're using Windows then you will need an editor which supports UNIX-style line endings (so, not Notepad).

Riffer, 2017-07-02:
Thank you very much. Was just able to test and it works as you described.

Issue can be closed...

closed 2017-07-02 by Riffer