opentx/opentx #5072: Unable to flash XM+ version request failed

sensei28, 2017-07-02:
Look at https://www.rcgroups.com/forums/showthread.php?2870872-Unable-to-flash-XM-version-request-failed

same problem

3djc, 2017-07-02:
We just confirmed it working on a 2.2.0 X9D+ anjd XM+, flashing did not produce and error and went fine. It must be something with your setup.

closed 2017-07-02 by 3djc

sensei28, 2017-07-04:
my setup is not wrong. i received "firmware update error: version request failed"

kilrah, 2017-07-04:
We did try it with an XM+ and our radios and it works, so it's not an OpenTX problem. Either you've done something wrong or your receiver is defective.
You did not include any details likey what radio you're using so even if we wanted we can't help further.

sensei28, 2017-07-04:
i am using taranis x9d+ and open tx 2.2.0. Receiver is a brand new receiver. After getting this error, i cant bind receiver. But before flashing i did.

sensei28, 2017-07-04:
i can record a video if you want

kilrah, 2017-07-04:
No point - but as said it's nothing we can do anything about.

sensei28, 2017-07-04:
i tried 2 brand new receivers and same error. Is it possible that receivers are fake. i bought from banggood.

kilrah, 2017-07-04:
Why are you trying to flash them in the first place?

sensei28, 2017-07-04:
i want to get rssi on channel 16 at betaflight.

kilrah, 2017-07-04:
Well, the only thing we can advise is to contact your seller and get help or a replacement from them.

sensei28, 2017-07-04:
ok ty so much for your help